bambulab/BambuStudio #195: Camera not loading: Loading Failed [0]!

TMeeib, 2022-09-02:
Hello together,

the camera feature on windows is not working for me. My guess is, that the issue is not the printer, because the camera works in the mobile app.

If i press play, i get the following error message from Windows Media Player
[image]
 Sorry for the german^^. It basically means, that the file has an invalid file extention (.) and it cannot play a file of this type.

If I answer "yes" to the question if I want to try it anyways I get a "Loading Failed [0]!" in the bottom right corner of the camera view

bambu123, 2022-09-06:
There is a windows registry key that affects this system dialog. Sometimes the system clears the registry key, such as a system update.
You can just check the checkbox in this dialog to avoid it, or reinstall BambuStudio to create this registry key again.

For the second problem, please try again after solving the first problem.
Please send us the log files in "%APPDATA%\BambuStudio\log" if it keeps failing.

TMeeib, 2022-09-06:
The checkbox suppressed the error message, but the camera is still not loading. 
`2022-09-06 12:00:00.510983[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Initializing...
2022-09-06 12:00:01.689002[Thread 0x000051ac]:camera_url: bambu:///TXBUHEU29B7UUD9G111A?authkey=77aia9zs&passwd=ux0zfj&region=us
2022-09-06 12:00:01.689002[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Loading...
2022-09-06 12:00:01.690003[Thread 0x00000b7c]:MediaPlayCtrl: start load
2022-09-06 12:00:01.698004[Thread 0x00000b7c]:MediaPlayCtrl: end load
2022-09-06 12:00:01.714019[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: -1, last_state: 5
2022-09-06 12:00:03.880612[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-09-06 12:00:03.880612[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: size: 0x0
2022-09-06 12:00:03.881613[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Stopped.
2022-09-06 12:00:03.883614[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Loading failed [0]!
2022-09-06 12:00:03.887618[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
2022-09-06 12:00:06.040025[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Initializing...
2022-09-06 12:00:06.704004[Thread 0x00001b10]:process_task{ type:Backup, id:0, path:, object:0, extra:214}
2022-09-06 12:00:06.704004[Thread 0x00001ca4]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:214}
2022-09-06 12:00:06.761033[Thread 0x00004a2c]:camera_url: bambu:///TXBUHEU29B7UUD9G111A?authkey=77aia9zs&passwd=ux0zfj&region=us
2022-09-06 12:00:06.761033[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Loading...
2022-09-06 12:00:06.762034[Thread 0x00000b7c]:MediaPlayCtrl: start load
2022-09-06 12:00:06.764036[Thread 0x00000b7c]:MediaPlayCtrl: end load
2022-09-06 12:00:06.770041[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: -1, last_state: 5
2022-09-06 12:00:06.937416[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-09-06 12:00:06.937416[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: size: 0x0
2022-09-06 12:00:06.937416[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Stopped.
2022-09-06 12:00:06.941419[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Loading failed [0]!
2022-09-06 12:00:06.944422[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
2022-09-06 12:00:09.911988[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Initializing...
2022-09-06 12:00:10.644987[Thread 0x00005ac4]:camera_url: bambu:///TXBUHEU29B7UUD9G111A?authkey=77aia9zs&passwd=ux0zfj&region=us
2022-09-06 12:00:10.644987[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Loading...
2022-09-06 12:00:10.644987[Thread 0x00000b7c]:MediaPlayCtrl: start load
2022-09-06 12:00:10.646990[Thread 0x00000b7c]:MediaPlayCtrl: end load
2022-09-06 12:00:10.653997[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: -1, last_state: 5
2022-09-06 12:00:10.830109[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-09-06 12:00:10.830109[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: size: 0x0
2022-09-06 12:00:10.830109[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Stopped.
2022-09-06 12:00:10.834112[Thread 0x00001ca4]:MediaPlayCtrl::SetStatus: Loading failed [0]!
2022-09-06 12:00:10.837116[Thread 0x00001ca4]:MediaPlayCtrl::onStateChanged: 0, last_state: 3`

I feel a bit uncomfortable to share the entire log here, can i send it as a Private Message? Or somehow restricted to project participants?

bambu123, 2022-09-07:
It seems that the BambuSource.dll was not correctly installed or missing. 
The BambuSource.dll lives in the directory '%APPDATA%\BambuStudio\plugins'.
Had you installed BambuStudio with a different windows account than the current working one?
Please reinstall BambuStudio for the current account.

TMeeib, 2022-09-07:
BambuSource.dll is in the '%APPDATA%\BambuStudio\plugins' directory of my user.
I have not changed anything. 
I do have one non admin account for day to day use and the system administrator account activated on my machine if i need elevated rights

bambu123, 2022-09-08:
Please execute the following in PowerShell and send the output here.

REG QUERY HKEY_CLASSES_ROOT\bambu\
REG QUERY "HKEY_CLASSES_ROOT\CLSID\{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}\InProcServer32\"

TMeeib, 2022-09-08:
```
HKEY_CLASSES_ROOT\bambu
    Source Filter    REG_SZ    {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}
```

```
REG QUERY "HKEY_CLASSES_ROOT\CLSID{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}\InProcServer32"
FEHLER: Der angegebene Registrierungsschlüssel bzw. Wert wurde nicht gefunden.
```
(Basically "Error: Reg Key not found")

bambu123, 2022-09-09:
It show that the BambuSource.dll was not correctly installed.
Please reinstall BambuStudio.
If it continues to fail, try installing BambuSource.dll manually to see what error occurs.

Step:

TMeeib, 2022-09-10:
That did the trick. I had to modify it a bit because %APPDATA% refers to the Current User and my Admin Account was not where the file was located. Thank you for your help.

regsvr32 C:\Users\Username\AppData\Roaming\BambuStudio\plugins\BambuSource.dll
where "Username" is your Username

So definitely an issue with my split of default user and admin for elevated rights. Might also be a problem for professional environments where you need an admin to install the program.

closed 2022-09-10 by TMeeib

TMeeib, 2022-10-22:
FYI, I have to re-run this command after every update. The solution is really easy but maybe you want to fix this someday.

bambu123, 2022-10-24:
The latest version of BambuStudio will pop up a dialog to require administrator permission to re-install BambuSource on this condition.