bambulab/BambuStudio #197: Mac scrolling is inverted in filament selection windows

drumbo, 2022-09-02:
When using Scroll Direction: Natural on a mac, if you edit a filament, the window scrolls correctly; as in if you use the trackpad "downwards", the page scrolls to the top. If, however, in that same edit window, you use the filament select dropdown to choose a different filament, the scrolling is now inverted - as in if you scroll "downwards" the page scrolls to the bottom of the window, whereas it should (like the previous window) scroll to the top of the list.

This actually happens also if you simply select a filament in the drop down in the mains screen - the scrolling is opposite to the way it should be

Mac M1 Pro using 12.5.1
Bambuslicer 01.01.01.01

bambu123, 2022-09-06:
Yes, it's a remaining BUG! I will fix it soon.

dereknheiley, 2022-09-09:
I thought it was just me and the scroll reverser app I used!

lanewei120, 2022-09-09:
already fixed in the newest released version v1.2.0

closed 2023-03-02 by SaltWei