bambulab/BambuStudio #198: Cut Tool-  Add Key option

schreiaj, 2022-09-02:
When cutting a model allow adding a key feature - a simple cube boolean cut from the model and then a key part added would be nice. You can already do this manually but having an option to just do it automatically and ensure the positioning would be convenient. 
[image]

schreiaj, 2022-09-03:
Expanding on the how after a bit of experimentation: 

The intersection of the cut plane and the model will generate a polygon. There's two approaches here - the first one involves simply finding the centroid of the polygon and placing the key there. The other option is computationally more involved but may result in more robust parts - picking a series of points within the bounds of the polygon and finding the top N (where N is configurable) based on some criteria like distance from edge and thickness of part at point.

zhimin-zeng-bambulab, 2023-08-18:
[image]
It has improved in the latest version, you can have a try.

closed 2023-08-18 by zhimin-zeng-bambulab