bambulab/BambuStudio #199: [Feature Request] Better preview when black filament is selected.

rwoldberg, 2022-09-02:
When previewing a sliced object and the filament you choose is black, it is impossible to see the lines in the output.  I would suggest limiting the preview to a minimum level of gray so this does not happen.
[image]
[image]

YifanWuBambu, 2022-09-03:
This is optimized and will be released soon.

arekm, 2023-11-17:
@Haidiye00 Could you make it even more brighter than it is in 1.8 beta? When viewing under small angles it is still to dark. (It is okish only when viewing perpendicularly or close to perpendicular).

Talking about commit https://github.com/bambulab/BambuStudio/commit/4a5bca24023e2e2e0a1d612dc5305f684a5e8d39

[image]

The model on screenshot is: https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/RockNLol/RockNRoll/STLs/rocker_left.stl

(duplicates:  https://github.com/bambulab/BambuStudio/issues/528 https://github.com/bambulab/BambuStudio/issues/2216 )