bambulab/BambuStudio #200: [Feature Request] Ability to pause the print a given layer

rwoldberg, 2022-09-02:
I would like to have the ability to pause the print at a given layer, this would give the ability to drop in parts while printing and have them embedded in the print.

ChrisG-hub, 2022-09-02:
This already in the software, from the first version. Just right click the green slider after slicing.
you really got to learn your slicer... :-)
[image]

rwoldberg, 2022-09-02:
Nice.  Not intuitive but that works.  Thanks.

closed 2022-09-02 by rwoldberg

ChrisG-hub, 2022-09-02:
Its been there in Prusa, Bambu, Superslicer and the original source where all these slicers been build on, but i can imagine coming from an other slicer it would be nice to explore and learn the ins and outs. Good luck printing. Its a hell of a machine.

rwoldberg, 2022-09-02:
I have primarily used Simplify3d.  Regardless of how many slicers it is in, it is still not intuitive.  Slider generally don't have right click functionality.   It is a hidden feature that you would never find unless you randomly start clicking on stuff. But I am glad it is there.