bambulab/BambuStudio #202: [Feature Request] Filament Profile Sharing (Community Contributed)

psiberfunk, 2022-09-02:
No slicer has implemented this idea yet, so far as I know, but it would be great if there was a way to aggregate, vote on , and rank , and then download slicer profiles for your favorite filament.

I imagine the way this would work would be similar to "community profiles" for game controller mappings in Steam, or some other in app, or web-linked repository where people could share their filament profiles and people could vote on the best ones.

The benefit of this would be massively opening up people's ability to use the printer in a turn-key fashion with a massive variety of filaments (assuming the idea/community takes off and people contribute).  You could incentivize people who get a certain number of votes for a well-tuned profile or are consistently at the top of the pile (good citizens) with a discount coupon for filament or accessories or something.. (prusa does this I think).

RobinWhitfield, 2022-09-04:
I think this could become a bit like Steam Controller profiles where people have stupid names for them and everyone just uses the top one (as either they just want to pick one without really knowing which one to pick, or they'll probably configure it themselves), but I would also like to see it happen.

I think for this to work well, it ideally needs a good way to compare filament profiles as well as a parametric filter. Users need to be given as much information as possible as simply as possible while they're looking to select a profile.

Names should also be standardised to stop things like: "_ESUN PLA-PRO SUPER FAST BEST QUALITY <3 <3 <3 ROFLCOPTER LIKE AND SUBSCRIBE #3DPRINTSFORLIFE_.json"

If this happens **it would** be a fantastic value add for the printer ecosystem.

AbeFM, 2022-09-09:
Perhaps Bambu could curate the list.