bambulab/BambuStudio #204: Colours cannot be assigned to filaments, only "slots".

RobinWhitfield, 2022-09-03:
I can assign a colour to a slot, but then when I swap filaments, this colour is retained.

The colour should be set for the filament selected (probably in the filament settings window rather than the main interface), not the slot itself.

White filament, set to white
[image]
Change to blue filament, colour stays white
[image]

YifanWuBambu, 2022-09-05:
@RobinWhitfield 
Color information is not a property of filament preset.
A filament preset is only related to a material type. If we introduce the color, there will be so many presets that we cannot enumerate all of them.

RobinWhitfield, 2022-09-05:
After posting this I realised that of course especially for multi-material you want to be assigning a colour to a slot rather than a filament type, but for those of us using specific specific filament rolls on a non-AMS system it would be great to be able to set a "default" colour as an optional field.

PrusaSlicer does allows you to do such a thing at present.

YifanWuBambu, 2022-09-08:
@RobinWhitfield 
"default" colour is a good idea. We will add this to our plan list.

YifanWuBambu, 2022-12-16:
This is included in v1.4.