bambulab/BambuStudio #209: Allow Mixed Material Printing

AbeFM, 2022-09-04:
I'm trying to print items I've printed a lot of on my Prusa MMU. I use PET-G as a breakaway support for my PLA prints. I've done this tens of times without issue.

BambuStudio won't allow me to slice. It says the temperature is the issue, so I adjusted temps until they were _identical_ but was still not allowed to use it.

Then I took the PET-G profile, selected "PLA" as the type and I'm allowed to print with the higher temps.

While the warning is good, not being allowed to use the printer is a big pain, now I have to make fake profiles, load the AMS with a different material than it really is, and generally it's hard to track what's going on. An option to acknowledge the warning and print anyways would be WAY more useful.

As a side note - I've yet to check bed temps, but it's important on MM prints to hold the bed temp at the first layer's temp the whole time - cycling the temps up and down can lead to failures.

YifanWuBambu, 2022-09-08:
@AbeFM 
This is in our plan list.

candre23, 2022-10-06:
Tried to print some PETG with support-W as an interface material, and BS won't allow it.  Even though there is plenty of overlap in the temp ranges for the two materials, and even after I manually specified to print both materials at the exact same temp, the "large temp difference" error persisted and it refused to slice.  

Ideally the software should be smart enough to look at the acceptable ranges for multiple materials and pick a temp somewhere within the overlap of those ranges (assuming overlap exists).  Failing that, it should recognize when manually specified print temps are identical (or at least close enough) and slice the model.  Failing _that,_ there should at least be some way to override the error and slice with the user-provided parameters, regardless of what imaginary problems it thinks exist.

AbeFM, 2022-10-12:
On other multi material machines, you'd pick a single bed temp for
everything (to prevent temperature cycling and part release), then the
nozzle temperature did what they do.

If you need a checkbox for "advanced" add it, but it's frustrating not
being able to fully utilize the machine.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2022-11-10:
Bump. Still an issue here. Just looking for a way to override the defaults here.

Thanks guys.

SaltWei, 2022-11-23:
Please wait for next release in Nov.

SaltWei, 2022-11-23:
Sorry, current is Nov. Should be next month