bambulab/BambuStudio #210: [bug] Start G-code causes hot nozzle to burn a hole in the bed.

RobinWhitfield, 2022-09-04:
In the start g-code there is the following line (107):
_G0 X135 Y253 F20000  ; move to exposed steel surface edge_

It then does a low precision homing command, permitting up to 300C on the nozzle (the nozzle is set to 20C below the initial layer temp a few lines up)

Y253 is not the exposed steel surface. It's on the bed. The exposed steel is Y261.

So this is presumably why the nozzle is burning a hole in the bed, as it's heating the nozzle super hot and then plunging it into the bed to home the printer.

Should it not be:
_G0 X135 Y261 F20000  ; move to exposed steel surface edge_

SaltWei, 2022-09-08:
In actual testing, it is found that if the user does not place the hot bed board properly, or put a non-Bambulab hot bed board without wiping the mouth and ears, there will be a problem when homing, which will lead to subsequent collisions. In order to eliminate this hidden danger, we finally decided to place the midpoint of the return on the edge of the hot bed printing area to ensure that the machine will not be damaged.

RobinWhitfield, 2022-09-08:
@SaltWei, that is interesting.

Can you clarify what you mean by the mouth and ears?

Should this low precision home be set to only permit a low nozzle temperature to ensure it will not affect the bed surface, as is done later for ABL?

SaltWei, 2022-09-08:
"non-Bambulab hot bed board without wiping the mouth and ears" should be "non-Bambulab hot bed board without wiping area".
Wiping area is shown below

[image]

SaltWei, 2022-09-08:
> Should this low precision home be set to only permit a low nozzle temperature to ensure it will not affect the bed surface, as is done later for ABL?

Going to comfirm and will reply later

RobinWhitfield, 2022-09-08:
Ahh, I understand.

I had not considered the possibility of that area not existing (either due to no bed, or a non-Bambu bed etc) and the issues that it could cause.

RobinWhitfield, 2022-09-09:
In 01.02.00.04 this section of start G-code is unchanged.

RobinWhitfield, 2022-09-13:
Still unchanged in 01.02.00.09

SaltWei, 2022-09-14:
lower temperature has no effect.

RobinWhitfield, 2022-09-14:
Ahh right, so is this just due to the force that the low precision home smushes itself into the bed, or is this hole coincidentally in this location and it's actually caused by something else (a pause before the back corner purge or something)?

*Something* is causing a hole to be put in the bed...

SaltWei, 2023-03-02:
Seem no update for a long time. Since printer firmware also changed a lot. Please let us known if still exist

closed 2023-03-02 by SaltWei