bambulab/BambuStudio #211: [bug] Slicer and AMS gets out of sync and prints with different slot

attilah, 2022-09-05:
Repro:
- AMS 1,3,4 is loaded with PLA
- Print a model, Slot 1 runs out of filament midprint, load some (same in my case) filament, resume printing, printing finished successfully
- At this point slicer shows a ? in Slot 1
- I hit print in slicer, and it prints the model with Slot 4 - I just noticed because of the color difference, haven't checked the slot assignment as it was a print from the next plate from the same project file.

How I see is the filament run out is remembered by the AMS and sent to slicer, but filament reload and the fact that the print finished successfully with Slot 1 (which means it has filament) is not communicated to the slicer.

Expected behavior:
AMS shows the last information (in case of Bambu Lab filament the correct information) and when printing the next model slicer:
- Either pops up a warning to the user that the printing slot is changed from the last print
- Persists the information that Slot 1 has filament, user does not have to do anything, next print starts from Slot 1.

attilah, 2022-09-07:
@YifanWuBambu probably related in the firmware, but, when I send the model from slicer to print from Slot 3, it does it correctly, but when I hit print again on the printer's touchscreen, it starts the print from Slot 1....which does not make sense, it should print from Slot 3 to correctly repeat the last job

YifanWuBambu, 2022-11-10:
> @YifanWuBambu probably related in the firmware, but, when I send the model from slicer to print from Slot 3, it does it correctly, but when I hit print again on the printer's touchscreen, it starts [...]

This issue should have been fixed in the latest firmware version. Please try to update the firmware.

sodachen22, 2022-11-28:
@attilah In the latest firmware, there is be a menu shows current AMS mapping, and the mapping can be changed when PRINT AGAIN on the printer's touch screen. Can this resolve you problem?

attilah, 2022-12-02:
Cool, thanks, I'll check. Please in the future add a version number instead of "latest" to make sure we're checking against the right one.