bambulab/BambuStudio #213: Layer shift on exported

Learoy1, 2022-09-05:
[image]
[image]
[image]
[image]

I've sliced and printed the models shown in the pictures and they printed fine but I also exported the G-Code /sliced file for transfer onto SD card and reprint later. 
I've reopened up the sliced files in Bambu slicer and the g code shows a layer shift with different colours.
It only appears to be on models with more than one colour.
I've not test printed the exported files yet to see if it does shift the layers.
Pictures show the shift on exported sliced code but I've also posted pictures of what it should look like/what it initially sliced.

YifanWuBambu, 2022-09-08:
@Learoy1 
Thanks for reporting this bug. We have reproduced it and will get it fixed.

Learoy1, 2022-09-08:
> @Learoy1 Thanks for reporting this bug. We have reproduced it and will get it fixed.

You're very welcome.

YifanWuBambu, 2022-09-13:
Fixed in v1.2.

closed 2022-09-13 by YifanWuBambu