bambulab/BambuStudio #214: Allow manual setting of pressure advance/linear advance, esp for dark filaments

weiliang90, 2022-09-05:
The flow calibration does not work properly on most black filament.

Would Bambulabs be able to expose the pressure advance setting to the user to set manually under filament settings?

I also suggest having the printer report the measured pressure advance setting, so that user can troubleshoot blobby/underextruded corners.

AbeFM, 2022-09-06:
I second this one. Pet specifically never calibrates, and blacks in general struggle. I would love to make it use another material bay or give the user the option to use the previous print's values.

psiberfunk, 2022-09-07:
It should definitely be possible to manually override the pressure advance value for a particular filament.. especially if flow calibration isn't working for some of them.  Auto-only with no manual override/way to run a classic Marlin/Klipper style calibration with lines and pick for yourself is a bummer.

I would widen this feature request to implementing something like a guided wizard to calibrate troublesome filaments by eye (with the lines method).  Then, at the end of the wizard, you would store that new PA value with a "don't try to auto-calibrate this filament" flag set in the profile, which would bypass the auto-calibration routine of the printer.

RobinWhitfield, 2022-09-08:
Can you not just add an M900 K0.## command into the filament start G-code for the filaments you want to manually adjust for?

Presuming this start G-code is run after the printer start G-code it should override even if you forget to turn off the calibration routine.

It would be great to make this more user friendly though, with some kind of manual calibration assistance.

There's also an M variable that's used with the M900 command when the printer sets to defaults, but I've not worked out quite what it's doing.

psiberfunk, 2022-09-08:
> Can you not just add an M900 K0.## command into the filament start G-code for the filaments you want to manually adjust for?
>

There are several problems with this hack approach although it might work:

1) The bambu labs ecosystem aims to be plug and play, this is antithetical to that ethos by forcing the user to get deep in the muck.

2) this makes assumptions about firmware support and flavor.  Do we even know if they’re running klipper or marlin, or if the firmware would respect this setting ? As far as I know there’s been very little open discussion about what the firmware flavor or source is. Please correct me on this if I’m wrong ?

3) such an unofficial approach is liable to break if Bambu changes their underlying firmware in a way that reorders commands

SaltWei, 2022-09-14:
If you want to manually set the flow calibration, just set gcode like "M900 K0.040" at the end of start gcode. 0.040 can be changed to other numerical value.

closed 2022-09-14 by SaltWei

psiberfunk, 2022-09-14:
@SaltWei is this actually fixed somehow ? I was thinking it should be fixed in the GUI , especially because there's no documentation saying M900 is respected or what the proper K value is.  I don't really think this ought to be closed with a comment that this should be hacked in via manual GCODE.

AbeFM, 2022-11-10:
I would be fine with a modifier... I have two cases where I can't get the control I need:

Some materials don't scan well with the camera. Black PETg, for instance. I would be ok with feeding the machine a stand in filament to record values from then apply it, but really, just an override would be fine.

The other case is NinjaFlex, which I think reacts slower than you test for. I think taking the measured number for your linear-advance-like number and adding 30% would fix my prints.

Basically the same as psiber said: Make it so we can work around it, or include it as a polished feature, but make it work. :-) Thanks.

nirurin, 2023-08-20:
Has there been any progress on this? Being unable to set pressure advance k values per-filament is pretty limiting, and it's a 'feature' orcaslicer has had for a long time

psiberfunk, 2023-08-20:
Their "response" is the bizarre AMS and machine-centric setting in the latest firmwares where it's somehow stored on the machine instead of in a silcer profile (where it really belongs).   This is totally bonkers IMO and is a convoluted, over-engineered and backwards UX design.  I have no idea why they did it this way instead of like every other slicer... but I'm guessing they'll tell you that's their solution.

My opinion, if you hadn't figured it out from the above.. is that this is not a good solution and is almost purposefully complex and locked to the machine itself.  I wish they'd throw it out and do what every other slicer did.. but Bambu can be really stubborn

nirurin, 2023-08-20:
> My opinion, if you hadn't figured it out from the above.. is that this is not a good solution and is almost purposefully complex and locked to the machine itself. I wish they'd throw it out and do what every other slicer did.. but Bambu can be really stubborn

Well, this isn't an opinion, it's just a fact. This solution does not work. It only allows you to set pressure advance for the very limited list of materials that are listed for the AMS, and you are completely unable to set pressure advance values for any other material or brand. 
Bambu may be satisfied with this, because they only want you to buy their own filaments, but it is terrible for their customers.

lanewei120, 2023-08-21:
> is, because they only want you to buy their own filaments, but it is terrible for their customers.

1. we are adding a way to create 3rd-filaments in the following versions
after doing this, you can also set it on the printer, and can save the values related with these filaments

2. and we stored the factor-K values on printer for following reason:
the compensation parameter is not only relative to the viscosity of the melted filament , but also relative to the melt zone diameter , nozzle diameter , nozzle length . As there is tolerance in the manufacturing , the melt zone diameter , nozzle diameter , nozzle length will not perfectly match the design , so the final compensation parameter for an specific filament will be different on different machines , and for now there is no a good solution to decouple these factors.

so we stored the calibration values(K factor) on the printer side, not in the filament preset.

It is also to ensure the versatility of the 3mf file. Since the parameters are related to the individual machine and the type of consumables, if the original user's 3mf has material parameters, it will cause other users to use the original user's parameters directly after slicing, which does not match the machine and their own materials, resulting in problems

psiberfunk, 2023-08-21:
> 2. and we stored the factor-K values on printer for following reason:
>        the compensation parameter is not only relative to the viscosity of the melted filament , but also relative to the melt [...]

This is true in theory in some limited way, but in practice, people regularly use similar K values for a given filament.  All the evidence you need for this assertion is evident in the popularity of [profiles like these](https://www.printables.com/model/455802-petg-tuned-x1c-print-profiles-for-bambuorca-slicer) which work pretty well for people despite having different printers.

Similarly, you could make arguments like this for fan speed, ideal line width, volumetric flow limits.. etc etc... and yet.. you do not.  The choice to set K values in the **printer** from your custom calibration workflow is utterly backwards from a design perspective.  You've placed a key parameter in a very inaccessible and hard to modify place with very little flexibility. 

The reality is that Bambu's non-linear K/L/M settings scanned from Lidar are effectively no better than using the marlin style K values used via with KXXX L1000 M10.   How do I know?  Because i've tuned many many filaments both ways, and the results are the same on a wide variety of PA tests.  The lidar also has a lot more jitter than a hand-tuned K value.  It's certainly more convenient , but the values should be stored to the filament slicer profile.. perhaps with some sort of nozzle-size/type-specific field or something... or just have users make a new filament entry for each nozzle size they wish to use it with.... but the absolutely worst way to design this UX is to store these values on the printer where they are difficult to find and share, hands down.