bambulab/BambuStudio #215: [Feature Request] Ability to assign specific color to infill or perimiter

Hudson-Protopasta, 2022-09-05:
I would love the ability to assign a specific AMS slot to the infill or perimeter doing so would allow for more dynamic models such as the ability to print a colored infill with a translucent shell. This feature is already available in the other slicers including in Prusa Slicer.

Duckshow, 2023-01-18:
I would also love this, especially since it would enable you to use a more expensive material on the skin of the model, and a cheaper material as infill.

JosefWN, 2023-09-21:
This could potentially apply not only to color, but also to other filament properties. Using regular PETG for infill but ESD PETG for all outer walls would significantly reduce the model cost for those of us who care about ESD, for example...

thenewguy, 2025-05-26:
Would be very useful for the H2D

whend3r, 2025-06-29:
We need this ASAP for the H2D, best with fine control for outer walls, inner walls and infill as well as top and bottom layers!

TheRanker, 2025-12-04:
Why is this not implemented yet?!? 

It's been years ... Come on bambu!!