bambulab/BambuStudio #216: Changing infill direction causes other parts to not print some layers

str8Astud3nt, 2022-09-05:
While experimenting with changing the infill direction to make the infill lines look nice I noticed that it was causing the other parts of the model to stop printing what appears to be their top and bottom shell layers.

[image]

However, this doesn't seem to happen when splitting the model to objects instead of parts.

[image]

But it does cause some other artifacts in the other objects, but that is unrelated to the infill direction bug.

[image]

YifanWuBambu, 2022-09-08:
@str8Astud3nt Could you help share your 3mf to us? Thanks!

str8Astud3nt, 2022-09-08:
Here is a google drive link.
https://drive.google.com/file/d/1V4EsmiAN_y-nu0SOr_osIgC7XOsvt9gZ/view?usp=sharing

Not sure the best way to share 3mf files since github doesn't accept that type automatically.

YifanWuBambu, 2022-09-08:
Thanks. We have reproduced this bug and will fix it soon.

closed 2022-10-12 by SaltWei