bambulab/BambuStudio #219: Add command console to manually execute g-code commands

isaaxa, 2022-09-06:
Requesting to have a command line console added to have the ability to manually execute g-code commands.

YifanWuBambu, 2022-09-08:
Hi @isaaxa ,
Could you help describe the specific senarios that you may use this feature?

RobinWhitfield, 2022-09-08:
I could see this maybe being useful for manually setting K values with M900 while testing, or if #225 is added, to send M117 messages to query values within a currently running print (if that's possible).

Definitely a feature for the tinkerers.

DocHogan, 2022-09-10:
> Hi @isaaxa , Could you help describe the specific senarios that you may use this feature?

I'm in a situation where I need it. Just finished a PC print, now attempting to print a PLA project. Fails to load PLA because it cannot purge PC. Cannot set temp to PC temp to manually purge.

a CLI would allow manual temp-load-purge

RobinWhitfield, 2022-09-11:
Surely the current manual temp and extruder control will allow you to do this purge?

bani6809, 2022-10-11:
simplify3d has a command console. octoprint has a command console. there are many good reasons to have one, and no reasons not to.

G4Cab, 2023-06-04:
+1 , a console for manual gcode control would be nice.
Integration in repetier-server, octoprint and so on would be nice too, but this may be not possible, cause the missing usb/serial port

duffpop, 2024-06-03:
I'd love to be able to run [cold pulls](https://forum.bambulab.com/t/cold-pull-g-code-for-us-lazy-bastards/54453) via the slicer, and I frequently find myself wanting to extrude x amount of a filament without the faff of nested menus. Accessing a console would make such tasks trivial.