bambulab/BambuStudio #221: [Feature Request] Customizable Adaptive Layer Height

seanjhseo, 2022-09-07:
Prusa Slicer has a feature in it's "Variable Layer Height" where we can select certain layers to have finer layer heights. 
E.g., 0.15 base layer height and selecting the top heights of my parts (dome shape) to gradually print to 0.07mm height. 

I see that BambuStudio has Adaptive Layer Height, but turning this feature on lets the slicer make automatic decisions on the layer height thicknesses. It would be great if we could selectively “paint” sections of our parts to print at a lower layer height, and also have the feature to blend the layer height differences.

I’m used to using Prusa Slicer and seeing how similar Bambu Studio is to PS, I was hoping that it would have this feature or plan to introduce it in the future. 
[image]

bani6809, 2022-10-24:
This is adjustable layer height feature from prusaslicer. It would be nice to have it in BS.

https://www.youtube.com/watch?v=sLOfYDy3Brk

rkarlsba, 2022-11-20:
I just stubled over this need - the code is in prusaslicer - could it be added, please?

empyrials, 2022-11-25:
This is a feature I really need for the products I make. Please add it & allow us to have fine controls over this

YifanWuBambu, 2022-12-03:
This will be included in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

astrof, 2022-12-17:
There used to be an adaptive layer height but I don't see it anymore?

YifanWuBambu, 2022-12-17:
@astrof 
Moved here
[image]

astrof, 2022-12-17:
But now it's on a per object basis and is not intuitive at all versus before was a global OR per object option (like in Cura or other slicers). I guess I'll have to mess with it more.