bambulab/BambuStudio #226: [Information Request] Please provide a page with the custom G-codes possible on this printer

RobinWhitfield, 2022-09-08:
There are quite a few custom G-codes on this printer, some of which are obvious in Bambu Studio, and some of which are not so clear.

Would it be possible to get a wiki page detailing the custom G-codes that can be used so that tinkerers can get the most out of this machine, in a similar way to Prusa details for their machines?

https://help.prusa3d.com/article/prusa-specific-g-codes_112173

RobinWhitfield, 2023-01-21:
Any updates on this?

It's ludicrous that we just have to guess at everything to try and use the printer to its full potential.

NairoDorian, 2023-01-21:
Yeah this guessing game is sad, please Bambu Lab help your users instead of making our life more complicated when you leave us in the dark, please provide full gcode documentation

MiscPrinter, 2023-01-21:
Updates from BL, no. I submitted a ticket asking for more information about the side bar light being controlled with gcode. That was 2 weeks ago without any response.

Gr4sil, 2023-03-03:
Hopefully there would be the listing of the g- and m-codes.