bambulab/BambuStudio #227: [Printer - bug(?)] When adding a custom filament into Bambu Studio, it appears as a blank box on the printer screen

RobinWhitfield, 2022-09-08:
When viewing a pre-sliced file on the printer, either one saved on a the SD card, or the last print sent over the network, if a custom filament has been set in Bambu Studio for the filament profile, this just appears as a green box on the printer LCD rather then showing what filament the file was sliced for.

In my case, the filament is POM, so I added POM in Bambu Studio. The printer clearly has no idea what POM is so I just get a blank green box.

For this to be possible the filament type box in Bambu Studio likely needs a short character limit setting too.

Adding more filament types to Bambu Studio (and the firmware) could presumably fix this too, but for future-proofing, auto-generation of filament type display on the printer would be better.

RobinWhitfield, 2022-09-08:
Example:
[image]

timothyzalusky, 2023-01-06:
I also would like to see custom filament names/types and brands, this could be really awesome for error proofing the support material that I use in the machine (stratasys SR30).