bambulab/BambuStudio #228: When low poly models are used, Bambu Studio/X1 seems to overcompensate causing unusual artefacts.

RobinWhitfield, 2022-09-08:
When a model is imported which is low poly so circles are not circles (and G2/G3 arcs are not being used), something seems to overcompensate on the shallow angle corners, causing an unusual and distinctive (and very obvious) surface finish.

I'm unsure whether this is an issue present on all shallow angle corners, or just when they're adjacent to a very short line.

Slowing down the print mitigates the issue to some extent.

This is not a general overextrusion issue as flow can be adjusted so that 90 degree corners are slightly underextruded and this issue is still present on curves.

Using a high poly model or a STEP file resolves the issue with curves.

[image]

[image]
[image]

3dprintandpaint, 2022-09-08:
[image]
Same problem here.  Left image is the Bambu X1C print, right is from a Prusa MK3S+.  High poly models are unaffected.

3dprintandpaint, 2022-09-08:
[image]

3dprintandpaint, 2022-09-08:
[image]

YifanWuBambu, 2022-09-12:
@RobinWhitfield @3dprintandpaint 
Could you share us the models? Thanks!

3dprintandpaint, 2022-09-12:
> @RobinWhitfield @3dprintandpaint Could you share us the models? Thanks!

Hi, this is a paid model, however I have sought permission from the owner and they have kindly agreed for the file to be shared.  Hopefully this helps.
[31_base_ring.zip](https://github.com/bambulab/BambuStudio/files/9545958/31_base_ring.zip)

YifanWuBambu, 2022-09-12:
@3dprintandpaint 
Thanks for your support. We will check this issue.

RobinWhitfield, 2022-09-12:
My specific model was: https://www.thingiverse.com/thing:1930493

psiberfunk, 2022-09-15:
This looks like an arc tolerance detection issue almost ?  I'm curious what the slice looks like in this case.  Do you guys have project files to share so we can make sure we're looking at exactly the same slice?

RobinWhitfield, 2022-09-15:
These facets are quite large so I wouldn't expect the printer to try and turn them into an arc.

The slice looked as expected - lots of straight lines making up the curves

psiberfunk, 2022-09-15:
> These facets are quite large so I wouldn't expect the printer to try and turn them into an arc.
> 
> The slice looked as expected - lots of straight lines making up the curves

So that's interesting, because when I used the default settings on your file , I got large arcs (although not the complete circle) like this:  
[image]

That's why i'm curious to see your project file.

RobinWhitfield, 2022-09-16:
Oh, interesting. Let me try again with a new version of Bambu Studio and see what happens

RobinWhitfield, 2022-09-16:
Nope, I still get lines on 01.02.00.09

[image]

[image]

psiberfunk, 2022-09-16:
> The main issue however is how the printer seems to be responding to a series of short lines, rather than whether or not arcs are being generated.

So, forgive my confusion here ..is the gripe that even when the gcode is little lines it the print should be smoother? … or is the gripe that the slicer should be better at generating arcs in these situations?  

or is the gripe that even when the code is generating arcs that the firmware is parsing that into small line segments that show up ? (I don’t think this last one is the gripe but I included this possibility for completeness )

RobinWhitfield, 2022-09-17:
The issue is that when the model is not being turned into arcs, the small line segments print poorly as there seems to be some kind of flow overcompensation. This is not an issue with most printers, where arcs aren't supported.

psiberfunk, 2022-09-17:
Got it, thanks for clarifying!

YifanWuBambu, 2022-09-18:
We are optimizing the vibration suppression algorithm.
For now, please slow down the outer wall speed to walkaround this issue.

H4kunaM4tata, 2023-01-08:
@RobinWhitfield  I am afraid this is going to get nowhere.
My ticket isn't as bad, https://github.com/bambulab/BambuStudio/issues/1006 , but here are examples I have had to deal with:

If I tell somebody this was printed by an AUD$1.8k (~USD1.2k) printer, they will call me an idiot for spending that much money for such quality

At this moment I am avoiding models that have low poly and upscalling SLA models which has higher quality to workaround this problem.
MK3S+ wasn't perfect but I never ever had such a terrible quality print like this

[image]

[image]

This problem is way beyond BambuLab wants to recognize since it also affect flat walls.
It took me 3 months, that is correct, 3 months and countless tests, countless filament wasted, countless f words thrown at the printer to get a print worth $1.8k

[image]

[image]

And yet, same model, same everything, few days apart.
On this case, the right model quality was affected because the CF rod requires almost everyday cleaning with isopropryl alcohol or it starts to create a wild resonance which is then transferred to the print.

[image]

It is safe to say I don't have a top quality printer but an Ender printer that requires me to constantly babysit it to print right.