bambulab/BambuStudio #231: [BUG] Nozzle push against the nozzle wiper during cool/heating cycle of first layer inspection

thrutheframe, 2022-09-08:
For some users, the nozzle pushes against the nozzle wiper during first layer inspection. This is mainly because the wiper is installed by humans on the factory grounds and there can be discrepancies. As the command for first layer inspection is hardcoded 3mf files that gets sent to the print, I would like to request that the nozzle moves a bit more to the left (1-1.5mm) to give a bit more space, just in case the installation of the nozzle wiper is not accurate enough.

thanks

JPotter73, 2022-09-19:
I second this issue. It appears the nozzle moves into the end of the wiper, pushing the plastic support and bending the wiper assembly 
[image]

It happens immediately after the first layer scan while processing. The normal placement of the nozzle during purge is not being used, it seems.

My first wiper assembly came pre-bent and I replaced it, The second is installed now, but is now bent also. The second layer sometimes drags filament into the print, that I can cut off after.
Thanks!

JPotter73, 2022-11-16:
I'd like to up vote this to a bug. My sock is being damaged by this bug. The head moves to the side of the screw on the wiper assembly at two points, I believe:

- After scanning the bed, it goes into the side of the screw to process
- After scanning the first layer, it does the same

Besides damaging the sock, this also means the wipe after completely misses the wiper and gets dragged out into the part.
[image]

JPotter73, 2022-11-19:
Another FB posting most likely from this bug:
https://www.facebook.com/groups/545438466970652/permalink/664698245044673/

djngnear, 2022-12-22:
Yup I just noticed my nozzle is wiping right inbetween the PTFE and the black plastic. I circled the spot in the pic. 

[image]