bambulab/BambuStudio #234: BambuStudio crashes upon starting it

jermbe, 2022-09-09:
I'm running OSX 10.15.3  (old Mac Pro), and BambuStudio crashes upon the final step of trying to install it (downloading of the plugin). 
If I uncheck the installation of the plugin it also crashes.
Now each time I try to open it, it always crashes.
I need to `rm -Rf ~/Library/Application\ Support/BambuStudio` in order to retry the install, but to no avail.

You do state that a minimum of OSX 10.15 is needed, so I guess there is something else wrong.
PrusaSlicer works perfect on the same Mac.

lanewei120, 2022-09-09:
thanks for reporting this issue

Is there any crash report?
(you can find it in the "Console" in a higher macos version)

jermbe, 2022-09-11:
Please find attached the carsh report
[BambuStudio_2022-09-11-102643.txt](https://github.com/bambulab/BambuStudio/files/9542076/BambuStudio_2022-09-11-102643.txt)

lhwei120, 2022-09-12:
the crash reason is like this:
Termination Reason:    DYLD, [0x4] Symbol missing

Dyld Error Message:
  Symbol not found: ___darwin_check_fd_set_overflow
  Referenced from: /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
  Expected in: /usr/lib/libSystem.B.dylib

currently I don't have a 10.15 to debug and we will try to find one.
could you use the find command to check whether there is such library on your computer or not?

find / -name "libSystem*"

if we can find one, then please use the command 'ln -s /the_found_path/libSystem.B.dylib /usr/lib/libSystem.B.dylib' to create a link and have a try?

jermbe, 2022-09-12:
I checked and that file is already present. I assume the symbol
"___darwin_check_fd_set_overflow" is not present in my version of the lib?

jermbe, 2022-09-14:
Hi, upon installing it, (no plugin installed), it seems to work: I can load an STL, slice it, and save it to my HDD.
I do notice the sliced file has the extension *.gcode.3mf haven't checked yet if the printers uses that extension to print.

But when I try to install the Bambu Network plugin, it crashes when the download reaches 50%, and then I need to `rm -Rf ~/Library/Application\ Support/BambuStudio`, in order to be able to reinstall it again.
In attachment the crash report
[BambuStudio_2022-09-14-111309.txt](https://github.com/bambulab/BambuStudio/files/9564785/BambuStudio_2022-09-14-111309.txt)

lanewei120, 2022-09-14:
[libBambuSource.zip](https://github.com/bambulab/BambuStudio/files/9566796/libBambuSource.zip)
please use attached libBambuSource.dylib for a try

remove the ~/Library/Application\ Support/BambuStudio/plugins/libBambuSource.dylib firstly
and then extract the libBambuSource.dylib from attached zip and copy it into that position
please do remove that file and then copy

jermbe, 2022-09-15:
hi, I had no  ~/Library/Application\ Support/BambuStudio/plugins/ folder,
as I am unable to install the plugin.
I manually created the plugins folder with the same permissions as all
other folders there, and unzipped the provided file there.
Upon starting BambuStudio it still says I need to install the plugin, and
when I try that it crashes again.

lanewei120, 2022-09-15:
Hello

then you need to do the steps after the crashes.

as there is another library libbambu_networking.dylib need to be installed
the behavior is in expect from your above steps

jermbe, 2022-09-15:
That seems to work, only remaing issue is that I can't seem to select my
Bambu printer. Although the app on my phone does detect my printer.
Any idea on how to fix that?
I'm on the same network

jermbe, 2022-09-15:
Seems I needed to login, which gave me another crash :-/

jermbe, 2022-09-15:
In crash report I see:

Dyld Error Message:
  Symbol not found: ___darwin_check_fd_set_overflow
  Referenced from: /Users/USER/Library/Application
Support/BambuStudio/*/libbambu_networking.dylib
  Expected in: /usr/lib/libSystem.B.dylib

lanewei120, 2022-09-17:
[bambu_plugins.zip](https://githucom/bambulab/BambuStudio/files/9590693/bambu_plugins.zip)
@jermbe 
please try these files with the same steps

this time, we need to remove all the dylibs in the plugins directory before copy them into it

jermbe, 2022-09-17:
Hi, when I do this, I keep getting the message that the plugin is not installed.
I did:
- clean install
- crash on plugin
- `rm *.dylib` in the plugins folder
- unzip the bambu_plugins.zip file

If I then retry to install the plugin it still crashes.

This is in the crash log:
`Crashed Thread:        0  bambustudio_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [7584]

Application Specific Information:
dyld: in dlopen()
`

lanewei120, 2022-09-18:
> Hi, when I do this, I keep getting the message that the plugin is not installed. I did:
>
> * clean install
> * crash on plugin
> * `rm *.dylib` in the plugins folder
> * unzip the bambu_plugins.zip file
>
> If [...]

Is there any information popup about the library?
and after the bambu_plugins.zip file unziped into plugins folder
could you give a snapshot of the libraries in that folder
and give the log in folder /Users/username/Library/Application Support/BambuStudio/log after the application bootup(crashed)?

thanks

jermbe, 2022-10-02:
Hi, sorry for the delay.

Attached are the logs and screenshot of the contents of the lib folder:
- [logs.zip](https://github.com/bambulab/BambuStudio/files/9692110/logs.zip)
- [image]
If you could tell me how I can compile the source on my Mac, I could debug is myself?

Kind regards,
Jeremy

lanewei120, 2022-10-02:
> Hi, sorry for the delay.
> 
> Attached are the logs and screenshot of the contents of the lib folder:
> 
> * [logs.zip](https://github.com/bambulab/BambuStudio/files/9692110/logs.zip)
> * [image]
>   If you could tell me how I can compile the source on my Mac, I could debug is myself?
> 
> Kind regards, Jeremy

it seems the log.zip only includes file with prefix "debug_network_xxx"
there should be some other log files like "debug_Thu_Sep_29_17_32_15.log.0"
which are necessary for debugging this issue

the compiling guide is here
https://github.com/bambulab/BambuStudio/wiki/Mac-Compile-Guide
however, currently your issue is about the network_plugin
not related with Bambu Studio

jermbe, 2022-10-02:
Hello, I'll try to the compiling guide later this week.
For the logs: tried it all again, and I get no other log files than the ones in the zip:
`jeremynaus@New-Manetheren log % ls -la
total 24
drwxr-xr-x   5 jeremynaus  staff   160  2 okt 11:06 .
drwxr-xr-x  11 jeremynaus  staff   352  2 okt 16:12 ..
-rw-r--r--   1 jeremynaus  staff  1868 17 sep 11:33 debug_network_Sat_Sep_17_11_33_01.log
-rw-r--r--   1 jeremynaus  staff  4096 17 sep 11:34 debug_network_Sat_Sep_17_11_33_57.log
-rw-------   1 jeremynaus  staff  2581  2 okt 11:06 logs.zip
jeremynaus@New-Manetheren log % `

lanewei120, 2022-10-02:
> Hello, I'll try to the compiling guide later this week. For the logs: tried it all again, and I get no other log files than the ones in the zip: [image]

just found that there is no more logs for the version run on old macos
so how about the full crash stack in the crash report now?

jermbe, 2022-10-02:
[BambuStudio_2022-10-02-161238_New-Manetheren.crash.zip](https://github.com/bambulab/BambuStudio/files/9692840/BambuStudio_2022-10-02-161238_New-Manetheren.crash.zip)
Here it is :-)

lanewei120, 2022-10-02:
it seems the library libBambuSource.dylib could not work well on macos 10.15
we will debug on it

as we are having our holidays this week, it will take some time.
Really apologize for it.

jermbe, 2022-10-02:
no worries, I'm actually amazed that you react on a Sunday :-)

jermbe, 2022-10-10:
I get the following error:
[image]

bambu123, 2022-10-11:
It seems everything is ok for the dylib to work on early macOS (10.15).

Could you please install the latest BambuStudio and the plugins? 
The Studio does not auto-update plugins, please remove the plugins in the plugins directory and restart studio.

The plugin's directory is located the same as the configuration directory which you can jump to from the menu "Help > Show Configuration Folder".

jermbe, 2022-10-13:
OK, I did all that, and it did not crash on installing the plugins, BUT upon trying to login the software crashes.
See crash report below
[BambuStudio_2022-10-13-101131_New-Manetheren.txt](https://github.com/bambulab/BambuStudio/files/9772351/BambuStudio_2022-10-13-101131_New-Manetheren.txt)

lanewei120, 2022-10-13:
> OK, I did all that, and it did not crash on installing the plugins, BUT upon trying to login the software crashes. See crash report below [BambuStudio_2022-10-13-101131_New-Manetheren.txt](https://github.com/bambulab/BambuStudio/files/9772351/BambuStudio_2022-10-13-101131_New-Manetheren.txt)

this crash file seems incorrect,
could you double check?

jermbe, 2022-10-13:
I thought the .crash extension was just a .txt :-).
I zipped it, it should now be readable :-), sorry
[BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip](https://github.com/bambulab/BambuStudio/files/9772404/BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip)

lanewei120, 2022-10-13:
> I thought the .crash extension was just a .txt :-). I zipped it, it should now be readable :-), sorry [BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip](https://github.com/bambulab/BambuStudio/files/9772404/BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip)

the crash file in the zip seems not a correct one, it is only 1 KB
it should be wrong somewhere

could you double check?

jermbe, 2022-10-13:
When I open it, I see all the following:
[image]

bambu123, 2022-10-13:
> I thought the .crash extension was just a .txt :-). I zipped it, it should now be readable :-), sorry [BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip](https://github.com/bambulab/BambuStudio/files/9772404/BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip)

The zip contains a symbolic link file, not the real one.

lanewei120, 2022-10-13:
it seems the libbambu_networking.dylib is still not working

please also help tp package the directory of /Users/username/Library/Application Support/BambuStudio
and send back it to us

thanks

jermbe, 2022-10-13:
Here you go, I excluded the user folder as I belief the config file contain some personal info like my userid etc

[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/9776621/BambuStudio.zip)

lanewei120, 2022-10-13:
> When I open it, I see all the following: [image]

please also send the file of this crash stack to us

as @bambu123 said, previously maybe a link file was sent?

jermbe, 2022-10-13:
> When I open it, I see all the following: [image: Schermafbeelding
> 2022-10-13 om 12 26 12]

dear, it is in the ZIP file I sent earlier. On a Mac, just double click on
the .crash file and it should open and show it fully :-)

lanewei120, 2022-10-13:
[image]
it shows a link to the file /Users/jeremynaus/Library/Logs/DiagnosticReports/BambuStudio_2022-10-13-101131_New-Manetheren.crash
I can not open it here

jermbe, 2022-10-13:
ah, so that's why I could open it :-)
Here is (hopefully) the correct one
[BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip](https://github.com/bambulab/BambuStudio/files/9776887/BambuStudio_2022-10-13-101131_New-Manetheren.crash.zip)

lanewei120, 2022-10-13:
[networking.zip](https://github.com/bambulab/BambuStudio/files/9777371/networking.zip)
please try this libbambu_networking.dylib

close the BambuStudio
remove the file /Users/username/Library/Application Support/BambuStudio/plugins/libbambu_networking.dylib
and copy the file in this zip into /Users/username/Library/Application Support/BambuStudio/plugins/

jermbe, 2022-10-14:
YES! :-) That seems to work, I've just sent a print from my Mac towards the printer :-)

THANK YOU ;-)

lanewei120, 2022-10-14:
great to hear that!

we will release these libraries soon,
and then you can just intall it from the network

lanewei120, 2022-10-14:
> YES! :-) That seems to work, I've just sent a print from my Mac towards the printer :-)
> 
> THANK YOU ;-)

we have uploaded the library to cloud
you can remove the plugins and let it be installed from internet

please help to have a try to confirm this issue to be closed or not

thanks

jermbe, 2022-10-15:
I just removed the plugins, and upon starting BambuStudio the app automatically starts to download the plugins. It all goes perfectly without a problem :-)

Thank you very much :-)

YifanWuBambu, 2022-10-21:
@jermbe 
Please also verify on the latest release v1.3.0.12.

jermbe, 2022-10-31:
Hi,

I just installed the update and it all works perfectly. Thank you, For me
the ticket can be closed :-)

Kind regards

closed 2022-11-03 by YifanWuBambu