bambulab/BambuStudio #236: Building on Manjaro Linux Failed to ArcFitter.cpp

lord-x-type, 2022-09-09:
Hi There.
i Tried to compile Bambu Studio on latest Manjaro Linux and getting this error:

`home/lord/BambuStudio/src/libslic3r/ArcFitter.cpp:52:99: Fehler: falsche Verwendung des unvollständigen Typs »class Slic3r::Polyline«
   52 |         can_fit = ArcSegment::try_create_arc(current_segment, target_arc, Polyline(current_segment).length(),
      |                                                                                                   ^
In Datei, eingebunden von /home/lord/BambuStudio/src/libslic3r/Circle.hpp:5,
                 von /home/lord/BambuStudio/src/libslic3r/ArcFitter.hpp:4,
                 von /home/lord/BambuStudio/src/libslic3r/ArcFitter.cpp:2:
/home/lord/BambuStudio/src/libslic3r/Line.hpp:15:7: Anmerkung: Vorwärtsdeklaration von »class Slic3r::Polyline«
   15 | class Polyline;
      |       ^~~~~~~~
/home/lord/BambuStudio/src/libslic3r/ArcFitter.cpp: In statischer Elementfunktion »static void Slic3r::ArcFitter::do_arc_fitting_and_simplify(Slic3r::Points&, std::vector<Slic3r::PathFittingData>&, double)«:
/home/lord/BambuStudio/src/libslic3r/ArcFitter.cpp:109:30: Fehler: unvollständiger Typ »Slic3r::MultiPoint« in geschachtelter Namensangabe verwendet
  109 |         points = MultiPoint::_douglas_peucker(points, tolerance);
      |                              ^~~~~~~~~~~~~~~~
/home/lord/BambuStudio/src/libslic3r/ArcFitter.cpp:131:48: Fehler: unvollständiger Typ »Slic3r::MultiPoint« in geschachtelter Namensangabe verwendet
  131 |             straight_or_arc_part = MultiPoint::_douglas_peucker(straight_or_arc_part, tolerance);
      |                                                ^~~~~~~~~~~~~~~~
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:76: src/libslic3r/CMakeFiles/libslic3r.dir/ArcFitter.cpp.o] Fehler 1
make[2]: *** Es wird auf noch nicht beendete Prozesse gewartet....
In Datei, eingebunden von /home/lord/BambuStudio/src/libslic3r/pchheader.hpp:117,
                 von /home/lord/BambuStudio/src/libslic3r/pchheader.cpp:1:
/home/lord/BambuStudio/src/libslic3r/Config.hpp: In Konstruktor »Slic3r::ConfigOptionEnumsGeneric::ConfigOptionEnumsGeneric(std::initializer_list<int>)«:
/home/lord/BambuStudio/src/libslic3r/Config.hpp:1648:14: Warnung: »Slic3r::ConfigOptionEnumsGeneric::keys_map« wird mit sich selbst initialisiert [-Winit-self]
 1648 |     explicit ConfigOptionEnumsGeneric(std::initializer_list<int> il) : ConfigOptionInts(std::move(il)), keys_map(keys_map) {}
      |              ^~~~~~~~~~~~~~~~~~~~~~~~
/home/lord/BambuStudio/src/libslic3r/Config.hpp: In Elementfunktion »bool Slic3r::ConfigOptionEnumsGeneric::operator==(const Slic3r::ConfigOptionIntsTempl<false>&) const«:
/home/lord/BambuStudio/src/libslic3r/Config.hpp:1661:13: Warnung: »throw« wird immer »terminate« aufrufen [-Wterminate]
 1661 |             throw ConfigurationError("ConfigOptionEnumsGeneric: Comparing incompatible types");
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [CMakeFiles/Makefile2:876: src/libslic3r/CMakeFiles/libslic3r.dir/all] Fehler 2
make: *** [Makefile:146: all] Fehler 2
`

Building the deps worked so far.. but now.. cant get over 15% of building the main Application..
Can You help me further with this?

lhwei120, 2022-09-12:
> Fehle

it seems the compiling output are not in English, 
could you switch to english and give the output here?

lord-x-type, 2022-09-13:
Hi, ok... no problem :) 
seems there are a lot of additional errors... but one after another.
here is the error on ArcFitter.cpp

[image]

lord-x-type, 2022-09-13:
here is the complete build error log for now:
[build.log](https://github.com/bambulab/BambuStudio/files/9554808/build.log)

lord-x-type, 2022-09-18:
Now.. the errors are gone (for good? ;) ).. i repeated cloning the repo and build the deps.. worked as last time.. 
then building the main application worked so far.
Starting the main app failed to not existing network plugin it seems.
so this looks like an other question like mentioned here #12 
Thanks alot.. this one can be closed.

closed 2022-09-21 by SaltWei