bambulab/BambuStudio #239: Embed Font Face in Drop-Down

ChrisPirillo, 2022-09-09:
The new text tool is great, but not as usable as it would be if you could see the font face rendered inline to know what it looked like before wasting time placing it into the model only to realize it's not what you wanted. As it stands, it's challenging to use unless you know the font face / style by memory. 

[image]

Separately, I have a locally-installed font (Windows 11) but it's not being pulled into this list in alphabetical order (anywhere)?

YifanWuBambu, 2022-09-12:
Showing font face in drop-down is in our plan list.

Separately, I have a locally-installed font (Windows 11) but it's not being pulled into this list in alphabetical order (anywhere)?
>> Please share the path where you install the font. Thanks

ChrisPirillo, 2022-09-12:
C:\Users\Chris\AppData\Local\Microsoft\Windows\Fonts

ChrisPirillo, 2022-09-12:
And, for reference, if I can attach the file... here it is. A regular ol' TTF, installed the normal way in Windows 11.

ChrisPirillo, 2022-09-12:
[Starjedi.ttf.zip](https://github.com/bambulab/BambuStudio/files/9550104/Starjedi.ttf.zip)

Here's a copy of the font, for reference - installed in the normal way for Windows 11 a while back (and it works in other applications on the computer, so I know it's available).

ChrisPirillo, 2022-09-13:
And, if you're going to implement the inline font-face display, I would also strongly suggest two other options that would make this option more useful:

* Favoriting fonts to the top of the list
* Recently used fonts
* Search for font (not just by first letter)

Photogad, 2023-03-27:
@YifanWuBambu 
I would also suggest a slider to adjust text size instead of having to type in decimals. Another suggestion may also be that text on objects has a slightly different color or highlight than the filament color, so the text can be visible to the user when they are viewing the model from above.

YifanWuBambu, 2023-03-27:
> @YifanWuBambu I would also suggest a slider to adjust text size instead of having to type in decimals. Another suggestion may also be that text on objects has a slightly different color or highlight [...]

Hi @Photogad ,
We are about to close this ticket. Could you please create a new ticket for these suggestions? Thanks a lot!

LiZ-Li-BBL, 2023-03-27:
You can now preview each font style on combobox, and searching a specific font is supported. Also, some custom installation fonts may not load into BambuStudio properly, you can keep an eye on this :https://github.com/bambulab/BambuStudio/issues/818
We will fix the problem as soon as possible.

closed 2023-03-27 by LiZ-Li-BBL