bambulab/BambuStudio #241: Spiral mode different buils plates or objects

coat64, 2022-09-10:
When setting spiral vase for one buildplate or object, the other plates/objects also changes to spiral vase. It is possible to set the other objects back to nwalls etc. afterwards. In other words, it doesn't seem possible to set one object (sequential printing) to vasemode without changing the other objects.

SaltWei, 2023-09-04:
We will make spiral mode as plate config in future version. Thank you very much!

michelem09, 2023-12-05:
Please this would be great.

SaltWei, 2023-12-06:
> Please this would be great.

Under developing

Evil-Error, 2024-11-25:
I wonder if we talk about to make it available "just" per  "plate" or also "per object" ? 
It would be usefull and I guess no problem if I could setup "print order per object" and then configure one object "normal" and the other one in "vase-mode". 

btw. that request is now 1 year old. Is it still planed to implement this feature?