bambulab/BambuStudio #243: Support painting camera out of sync with other viewport (prepare/preview) cameras

kurris19, 2022-09-10:
SUMMARY

When you click to start painting supports, the viewport camera position seems to have its own independent position -- it does not match the camera position of either Prepare or Preview.

STEPS TO REPRODUCE

1. Load a model, aim the camera in some fashion that zooms in close, or is some interesting angle
2. Click on the "paint supports" button
3. Camera is somewhere completely different

EXPECTED RESULTS

The support painting camera should always be the same as the Prepare/Preview camera.

YifanWuBambu, 2022-09-12:
@kurris19 
What is your studio version?

kurris19, 2022-09-12:
The latest, built from GitHub - 01.02.00.04, though it has been present for quite some time.

YifanWuBambu, 2022-10-19:
@kurris19 
Sorry for the late response. We will fix it in recent version. It actually brings some bad experiences.