bambulab/BambuStudio #244: Disable auto-play vid when switching to machine tab

DocHogan, 2022-09-10:
v1.2 has changed behavior to automatically play video when switching to Machine tab. On some PC systems (*) BambuStudio will crash when attempting to start video. Please disable video automatically starting when switching to Machine tab; preferably, make Autoplay a toggleable option (disabled by default).

(*) I experienced this crashing issue on a laptop with both intel integrated graphics and nvidia gpu; it took me some time to realize that this was the issue and manually assign BambuStudio to use the intel graphics controller.

bambu123, 2022-09-13:
@DocHogan Would you please help us to find the crash point? Can we talk with discord? My account is chunmao.guo#3806.

bambu123, 2022-09-13:
We use Windows Media Player for live view which uses the d3d9 driver for media renderer. 
From my experience, the D3d9 has some problems with some video cards on a dual image card system. 
But I can't tell the reason, and switching to another video card may work.

We had seen a hung problem with this issue, but haven't seen the crash problem. So we strongly need your help.

DocHogan, 2022-09-13:
"Hung" probably is a better description than "crashed", though the app never recovers and must be killed via End Task.

I can try to find opportunity to assist via Discord., though I have severely limited access; I do not work in an environment that I can use phone/pc during work hours, and I work 14 hour days, leaving me about an hour each day to do all personal tasks. I can schedule a time to be available, though.

bambu123, 2022-09-14:
Ok, if it is hung on, that is the same issue we are facing.
We are preparing to refactor the player and have to use d3d9 before the new one.

bambu123, 2023-03-02:
The auto-play issue is already fixed. So close.

closed 2023-03-02 by bambu123