bambulab/BambuStudio #245: [FEATURE REQUEST] Better camera/viewport navigation

kurris19, 2022-09-10:
Bambu Studio would significantly benefit from more robust zoom/pan behavior.

1. Zooming to the mouse pointer, instead of just the center of the viewport would be a HUGE advantage and make navigating the build plate much easier.
2. Providing an option to invert mouse/trackpad zoom direction would be great as well (most, if not all, 3D applications have this option, since some of us are backwards).
3. Pan seems to have its axis fixed on the camera, instead of orbiting the center of the viewport.  While this behavior is fine at a macro level, when you are attempting to visualize fine details at a high zoom level, it can be next to impossible to keep what you want to look at in the viewport without an incredibly tedious sequence of pan/dolly/pan/dolly/zoom/pan/dolly ("dolly" may not be the right word...track?...right-button move?...whatever).

RobinWhitfield, 2022-09-10:
Zooming to mouse pointer does make navigation on a trackpad more difficult  (I always end up using a mouse when in SOLIDWORKS), but it would bring it inline with other 3D software.

kurris19, 2022-09-10:
I don't know...I've used a trackpad exclusively with 3D software for 10 years or more and have no difficulty navigating with it.  Not as easy as a mouse, I suppose, but...[mental shrug]

warpster, 2022-09-20:
Not sure if this is covered here or not, the OP mentions PAN but my concern is Orbit (I think this is what the OP was referring to on #3. When zooming in on a model the Orbit is still centered on the center of the build plate making Orbiting almost impossible. It would help to have the center set to the center of a model if a model is selected, otherwise set it to build plate (viewport) center.