bambulab/BambuStudio #249: In Color Painting, Clicking Other Tool Type Applies Existing Type to Model Behind It

kritoke, 2022-09-10:
Sometimes, especially when zoomed in painting the model, it will apply the current color painting type to the model instead of the button clicked in the tool type area.  Mostly having the problem when Fill is selected.  Having the issue on Studio version 01.02.00.04.

YifanWuBambu, 2022-09-12:
Hi @kritoke ,
I verified the issue on both windows (11) and mac(12.5), but did not reproduce this issue.

If possilbe, could you help record a gif (with ScreenToGif tool) or video?
Besides, what is your operation system version?

Thanks!

kritoke, 2022-09-13:
I already downgraded to the other version of Bambu Studio without the bugs, it doesn't really appear to be an issue on the previous version.  So probably was a regression.  I am running Windows 11.

YifanWuBambu, 2022-10-14:
Hi @kritoke ,
Do you have the issue on 1.2.0.9?

YifanWuBambu, 2022-12-03:
Close the issue for now.
If it is reproduced, please create a new one.
Thanks!

closed 2022-12-03 by YifanWuBambu