bambulab/BambuStudio #250: Auto Populate filament from AMS to slicer.

randyre, 2022-09-11:
Every time I reload/restart the slicer, I have to tell the slicer that there are four slots and then have to pick each one to get them in the right order., It already knows what's there, but it's not automatically populated.  This will only get worse when we add even more AMS units, I would hate to do this for 16 slots every time that I start the slicer.

In the files I uploaded you will notice that one only shows a single slot, the second shows all four slots but with random colors, and the third shows that I have to pick slots and the order for all four slots.

[image]
[image]
[image]

YifanWuBambu, 2022-09-12:
Maybe we can add a button to read AMS slots and replace the the filament list by filaments in AMS.

randyre, 2022-09-13:
Thanks. Enhancement label is good until AMS hub is available. If not added by then it should be upgraded.  Also woyld prefer auto population instead of a button, the slots are already read as proof in the pull down menus, not sure why it doesn't  just automatically populate. I don't see the reasoning behind it just sitting there with one slot and the wrong filament. Seems counter intuitive.

YifanWuBambu, 2022-09-14:
@randyre 
In Bambu Studio, project filaments are independent with AMS spools.
We do this for the following reasons:
1. When loading an existing multi-material 3mf project, its filaments are likely to be different with the spools in the AMS. We want to keep the original filaments used by 3mf project.
2. You may create a project that uses more colors than your AMS slots, and schedule them to serveral plates.

warpster, 2022-09-27:
So why not load the loaded into the AMS filaments if you are starting a new project? Sure if you load an existing one it should keep what was there but on startup it seems, since it is a new project it should populate.

YifanWuBambu, 2022-10-09:
@warpster 
We added an button to auto-sync the AMS filaments to current project (especially useful for new project).

randyre, 2022-10-21:
The newest (1.03.00.12) version has this button. Works great. Thanks.

closed 2022-10-21 by randyre