bambulab/BambuStudio #253: Difference prusaSlicer

MarcoPiazza83, 2022-09-11:
I don't understand why the slicing algorithm is different.
As you can see in this model, it is not a setting problem for some parameters, but the different calculation of the slicing engine. 

[image]

RobinWhitfield, 2022-09-13:
> I don't understand why the slicing algorithm is different.
Because they're not the same slicer?

If you switch to the same infill type I presume there's no difference?

MarcoPiazza83, 2022-09-13:
> > Non capisco perché l'algoritmo di affettatura sia diverso.
> 
> Perché non sono gli stessi affettatori?
> 
> Se passi allo stesso tipo di riempimento presumo che non ci sia differenza?

Left bambu studio, right prusaslicer.
No, look at how the geometry of the letters is created, completely different!

RobinWhitfield, 2022-09-13:
I'm aware - they're not the same slicer, and they do things differently.

It does look like it might be a bug though - can you replicate the issue of these permiters going inside the model in a simpler model?

If you lower the precision in Bambu Studio, does the issue go away?

SaltWei, 2022-09-19:
Could you share the model for us to check?

SaltWei, 2022-09-26:
Could you share a 3mf for us to check and compare?

SaltWei, 2022-12-16:
Close this ticket for no response for long time.
Please report issue ticket if have new issues.
Thank you very much for suggestions

closed 2022-12-16 by SaltWei