bambulab/BambuStudio #254: Add Aligned Rectilinear Infill (and rename Zig Zag to Rectilinear)

skysyk, 2022-09-12:
Please add the Aligned Rectilinear infill option from PrusaSlicer that enabled infill lines to be closely matched to each other throughout the sliced model.

This is a useful option for certain 100% infill applications such as printing solid transparent objects.

Also requesting Zig Zag be renamed Rectilinear for consistency, but this is a nitpick.

RobinWhitfield, 2022-09-13:
I expected this request following the release of a certain transparent object video!

SaltWei, 2022-09-21:
> Please add the Aligned Rectilinear infill option from PrusaSlicer that enables infill lines to be closely matched to each other throughout the sliced model.
> 
> This is a useful option for certain 100% infill applications such as printing solid transparent objects.
> 
> Also requesting Zig Zag be renamed Rectilinear for consistency, but this is a nitpick.

Done. Please wait for next release.

closed 2022-09-21 by SaltWei