bambulab/BambuStudio #255: Make colour change gradual from dark to light (and back)

mctazman, 2022-09-12:
When printing in 3 colours (or more), there of course a colour changing algoritm. In my corrent print it's: eSun Bone White (which is yellowish) -> eSun Firetruck Red -> BL White -> eSun Firetruck Red -> eSun Bone White (which is yellowish)

Admittedly, I've got my purhe levels way down, but this order makes sure that all parts have red in them. In this case the best order (based on my colour input on the AMS) should be:

BL White -> eSun Bone White -> eSun Firetruck Red -> eSun Bone White -> BL White

That order minimizes the colour bleed, as it goes from dark to lighter to light to lighter to dark.

mctazman, 2022-09-12:
[image]

Here's the current print's colour settings. The White PLA's BambuLab, so that's RFID'd. The Red & Beige are manually set colours. So in this case the best order would be: Slot 1, 3, 2, 3, 1, etc.

YifanWuBambu, 2022-09-12:
Thanks. That's a good idea.
We will add this optimization in the future release.

YifanWuBambu, 2022-10-21:
Added in v1.3.0.12.

closed 2022-10-21 by YifanWuBambu