bambulab/BambuStudio #256: [FEATURE REQUEST] Minimal Read Only REST API

labatt, 2022-09-12:
- - To streamline project work and notifications across multiple printers and multiple jobs, it would be helpful if we could retrieve a minimal amount of information from the Bambu Labs Cloud:

- Printer status (i.e. Offline, Idle, Printing, Error)
- Temperatures (Nozzle, Bed, Chamber)
- Times (Print start, estimated minutes to completion for current print)

As a bonus, the ability to pick up the camera stream in some format... RTSP?

That's all that's REALLY important. Additional ones that would be NICE but not really as important:

- Speed %
- Printing Progress Step (i.e. heating, calibrating, etc.)
- Fan status (part cooling fan on, aux cooling fan on)

And when we get to two way...

- Stop print
- Pause print
- Run calibration

The ability to submit a print job would be great, but an MVP/iterative approach to get _some_ status info to be read about the printer would be amazing.

YifanWuBambu, 2022-09-13:
Hi @labatt ,
What is the purpose of minimal amount of information?
Save on network traffic or to display multiple printers/jobs at one screen?

labatt, 2022-09-13:
It's to integrate into a display that shows the status of multiple printers - x1, prusa, ender, etc. So we can see what printers are available and which ones are busy (and for how long) and to make sure everything is ok with them. Right now we have to go into the bambu slicer or walk into a different room to see if we can send a new job to the x1, and how long it will be until we can send a new job. For the rest of our printers we can just go into an app that shows all of their statuses in one place. Does that make sense?

The video would be helpful but is less necessary. We mostly use that so we don't have to walk to the printer to see if there's an issue with the print. For example, last night we had an ASA print turn into spaghetti but didn't know until we went into the room and looked at the printer. We just don't want to have two different systems - one for bambu and one for everything else.

YifanWuBambu, 2022-09-14:
@labatt 
Thanks for the detailed explanation. Multi-printer monitoring is definitely an reasonable requirement.
We have also considered about this recently. This was added in our plan list.

labatt, 2022-09-14:
@YifanWuBambu please let me know if you'd like any other information or have any questions. Literally, a single endpoint could be constructed with a JSON response...

{
	"status": {
		"curent_status": "printing",
		"last_progress": "Heating Bed",
		"last_error": "Spaghetti detected",
		"speed": "100",
		"fan_part": "50",
		"fan_aux": "0"
	},
	"temperatures": {
		"nozzle": "240",
		"bed": "45",
		"chamber": "42",
		"units": "C"
	},
	"times": {
		"print_start": "2022-09-14T18:59:02+00:00",
		"print_estimate": "5536"
	}
}

Speed is printer speed in %. The fan numbers are %. The print estimate is in seconds. etc.

YifanWuBambu, 2022-09-18:
> @YifanWuBambu please let me know if you'd like any other information or have any questions. Literally, a single endpoint could be constructed with a JSON response...
>
> ```
> {
> 	"status": { [...]

"print_estimate" means the left printing time?
Maybe "print_progress" (unit %) and "print_left_time" are enough?

labatt, 2022-09-20:
@YifanWuBambu print_estimate is the original time estimated for the print in seconds... and print_left_time in seconds would be good :) From print_left_time and print_start we can compute the print progress ourselves. And at the end, the last_final_print_time and last_filament_used would be helpful. We use "last_" on things because when we query printers, if the print is complete or a new job has started, the "last_" will still contain information about the job that just completed. But now I'm getting greedy. 

We do compare the time estimated in the slicer to the final print time in order to see how accurate the slicer is and make our own estimates when there are regular differences. And we track our spool usage so filament_used would also be helpful. Right now we're manually adding it to our system based on the slicer estimate.

labatt, 2022-09-20:
@YifanWuBambu I've updated the initial issue comment to reflect _everything_. Once again - I understand that not everything can be provided at first, or at all. But I just thought it would be good to keep track in the initial comment.

schreiaj, 2022-09-20:
An alternative approach could be to call a webhook at certain events with pertinent data. Initial thoughts on event list: 

CONNECTED - Printer powers on and connects
```
{
  printer_id: string of printer id currently displayed in app
  timestamp: timestamp
}
```

PRINT_STATUS|PRINT_STARTED|PRINT_ENDED - basically a generic printer status message, would be fired at start of print and either various percentages through gcode or frequency (I'd lean frequency, say every 5 minutes)
```
{
  printer_id: string of printer id currently displayed in app
  timestamp: timestamp
  print_information: {
    estimate: number of seconds estimated for print
    remaining: number of seconds remaining for the print 
  }
  printer_status: {
    extruders: [
        {
            temp: number degrees c for this extruder
            cooling_fan: number percentage cooling fan is at
        }
    ]
    bed_temp: number degrees c for the bed
    fans: [
        {
            percentage: number percentage of full this fan is at
        }
    ]
  }
}
```

PRINT_ERROR - on error

```
{
    error: string SPAGHETTI_DETECTION| FIRST_LAYER_INSPECTION | THERMAL_RUNAWAY...various errors 
    error_type: string TEMPORARY|PERMANENT - if this error can be recovered (eg FIRST LAYER inspection or SPAGHETTI) or if it is unrecoverable (eg THERMAL RUNAWAY)
}
```

jwalker55, 2022-09-21:
Would really like to see this as well. A REST API makes it really easy to integrate into things like Home Assistant too where you can setup your own custom monitoring/alerting.

Ali229, 2023-02-02:
Any update on this, I have found the handy app to be a hit or miss, would like an api to integrate into home assistant, send sms etc...

julie777, 2023-03-17:
Even better, I would like the 3D printing community to define a standard api for networked printers (with possible extensions for printers that do more or have more information).

Wouldn't it be great if you could start Bambu Studio and have it just see all your printers, regardless of brand, and be able to display some standard set of information and status, and accept print jobs from a queue?

Does anyone know of a group working to improve the 3D printing tool ecosystem to provide better interoperability?

For example, I think 3MF files could be a great transfer format if there was some standardization and well defined ways of providing specialized extension of contents.

Whether you choose to use a fully integrated environment like bambu studio, or have separate tools for configuring printers, slicing and managing a printer queue to send prints to one or more printers, the benefits of some standardized interfaces would be a huge improvement to the ecosystem.

jkenneydaniel, 2023-06-14:
Just wanted to provide some insight on this thread, there is a way currently to do this for X1 and P1P via MQTT. Each printer is running an MQTT relay, which you can connect & subscribe to the device report topic, and it will send you updates for the printer in JSON format including progress % and remaining time in minutes. I am using this to build my own dashboard to monitor printer status for all of my P1P's.

It's cumbersome, but if this is blocking you, it's definitely an option.

I am building out my interface via Go as a CLI to run 24/7 and will direct updates as they're received to a Laravel REST API where I can consume them and update my printers as needed, and render a grid on a large TV in my warehouse.

julie777, 2023-06-21:
I have played with the MQTT interface and I can monitor my P1P on my LAN easily.