bambulab/BambuStudio #258: [FEATURE REQUEST] - "Height Range Modifier"

cdsmakestuff, 2022-09-12:
is a feature that is available in PS and SS from an object right click context menu... allows changing of (printed) layer height(s) via a start layer variable and end layer variable independent of the global layer height setting. Is also object independent, modifier only effects object(s) selected, unselected object will still use global layer height throughout.

YifanWuBambu, 2022-09-13:
Hi @cdsmakestuff ,
Could you help provide some typical models for this feature?

ardenpm, 2022-09-17:
A strong use case for this functionality is decreasing the layer height on a specific part of the model to better print steep overhangs without pushing up the overall print time too high. [This model](https://www.printables.com/model/257851-giant-life-savers-container-with-dividers/files) for example makes use of this for exactly this reason (the included .3mf files for PrusaSlicer have that setup).

YifanWuBambu, 2022-09-18:
@ardenpm 
We will add "manual layer height" adjustment from prusaslicer recently.
That could sove this problem.

halofx, 2023-01-08:
The Height Range Modifier is a very powerful tool, that I use all the time! I desperately miss this feature from Prusa Slicer. I use it for layer heights, infill settings, perimeters, targeting ironing, and on, and on. Prusa Slicer allows so many settings to be targeted this way.

YifanWuBambu, 2023-03-02:
We've added this feature to our plan list. Thanks for your suggestions.

psy0rz, 2023-04-25:
closed in 1.6.0

closed 2023-07-21 by SaltWei