bambulab/BambuStudio #260: Bed texture gone missing on AMD systems

TheMrCaveman, 2022-09-13:
Prusaslicer and all other slicers as bambustudio do not work properly anymore on AMD systems.
Bedtexture is black and won't work. 

Please fix this.

In Prusaslicer this is already fixed.
See below!

Windows specific: Since AMD driver version 22.7.1, there are issues with the missing texture of the bed, possibly caused by a bug in the driver which seems not to accept texture mipmap generated on the CPU for some reason. To workaround the issue, PrusaSlicer checks driver version and if it is AMD newer than 22.6.1, it generates texture mipmaps on the GPU. https://github.com/prusa3d/PrusaSlicer/issues/8417, https://github.com/prusa3d/PrusaSlicer/issues/8508

lanewei120, 2022-09-15:
thanks for reporting this and the history information

we will fix it soon

danielclasen, 2022-09-23:
Yep, can confirm that is happening with AMD Adrenaline 22.1.9 as well:

[image]

lanewei120, 2022-09-23:
> Yep, can confirm that is happening with AMD Adrenaline 22.1.9 as well:
> 
> [image]

Is the version 22.1.9 or 22.9.1?
from the history of Prusa's issue, it seems this issue happens only on the driver version newer than 22.6.1

if the version is 22.1.9, could you also try PrusaSlicer to check whether this issue exists or not?

danielclasen, 2022-09-23:
It is 22.9.1, sorry for the typo!

danielclasen, 2022-09-23:
But I don't happen to have that Issue in Prusa Slicer 2.5.0

[image]

TheMrCaveman, 2022-09-23:
because prusa fixed it already. check the release notes for 2.5.0. This only happens if AMD driver is newer then 22.6.1. And yes I checked. It does not happen when driver is older then 22.6.1 with prusaslicer and bambuslicer.

lanewei120, 2022-09-23:
@TheMrCaveman 
thanks for the information
we will fix it in next release.

I checked the patch by prusa, it only fixed issues on version newer than 22.6.1
so I just confirmed the version is 22.9.1 not 22.1.9

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

closed 2022-10-21 by YifanWuBambu