bambulab/BambuStudio #262: [FEATURE REQUEST] Edit Text after placement

rzucco, 2022-09-13:
The new **add-text** feature is fantastic. However, there is no way to edit the text after you place it on the model.
Changing the characters would be a great time saver, as we could save it as a template instead of having to recreate it every time.

GuyH77, 2022-09-13:
2nd this. I came to add it myself but pleased someone else sees the value.

Manfredo1952, 2023-01-06:
Agree.  It is akward to remove and add the new text. Should be able to edit the initial insertion.- Thanks guys

mkosmo, 2023-01-07:
I'd like to voice my support for this feature request.

lemoanth, 2023-02-05:
I support the ability to edit text in the slicer

managerman, 2023-02-10:
Please add this feature!!  Thanks.

SaltWei, 2023-03-02:
[image]
Already available on v1.5.0 public beta.
But please attention that it's can't be edited after save and resave from 3MF again. This is hard to be supported.

closed 2023-03-02 by SaltWei

mkosmo, 2023-03-02:
I'd like to voice the opinion that I don't think this is actually resolved until it's editable after save.  This implementation is a stopgap rather than a solution.  I fully understand that you're operating within the confines of the 3MF spec, but there's surely a way to save the same metadata for later reuse after re-open.

SaltWei, 2023-03-03:
> But please attention that it's can't be edited after save and resave from 3MF again

@zhimin-zeng-bambulab Is this possible to be supported with limited resources?

kvnper, 2023-03-11:
Firstly, thank you for implementing this!

Secondly, I think that being able to edit after save would be very beneficial! For example, I make name tag keyrings and being able to save the project as a 3mf file and then come back to it later and change the name, would save me a lot of time customising the name if I could simply change the text.

sky-andrefaria, 2023-04-01:
I'll second @kvnper on this.

Opening up the project and edit the text is a huge time saver! And maybe avoid some manual mistakes on the way.

ysu2, 2024-07-13:
Another issue here is that the saved object does not carry any information as far as I can tell.
I cannot tell what font face & size was used to generate a text, so...

Yeah.  It would be a very-very well regarded piece of function if it was done. :)