bambulab/BambuStudio #264: Bambu Slicer Problem 2 mins to load up

P2012, 2022-09-13:
Hi i spoke to will @ (support) on your dicord and he told me to contact you. i have a problem using the slicer it takes at least 2 mins to load up will thinks it may be a network issue. i have no problems with other slicers. he told me to add the bambu config folder to help you

[image]

erickdredd, 2022-09-14:
Wild idea here, but I've had this issue with other slicers because of my Corsair keyboard. I'm not sure if this will help you, but try unplugging your keyboard and plugging it back in while the program is hung. This is something I've experienced with Prusa Slicer, Super Slicer, and Bambu. 

Example of where that issue was raised for PS: https://github.com/prusa3d/PrusaSlicer/issues/6050

P2012, 2022-09-14:
Thank you i will try that

P2012, 2022-09-15:
> Example of where that issue was raised for PS: prusa3d/PrusaSlicer#6050

Hi still the same thanks tho

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

YifanWuBambu, 2022-10-14:
@P2012 ,
Do you still have the same problem with v1.2.0.9 (latest released version)?

P2012, 2022-10-14:
Yes still takes about 2 mins to load

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

YifanWuBambu, 2022-10-16:
Could you send us the files under the configuration folder using 1.2.0.9?

Jump to the configuration folder by following steps:
Help menu => Show configuration folder

Just pack the whole folder as a zip to us.
It is helpful to find out the root cause.
Thanks a lot!

P2012, 2022-10-16:
Hi gmail will not let me send the file

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

YifanWuBambu, 2022-10-17:
@P2012 
Replying message on github, instead of gmail, could allow you to add the file.

P2012, 2022-10-17:
[System Configuration.zip](https://github.com/bambulab/BambuStudio/files/9798548/System.Configuration.zip)

Yes all done see file above @P2012

P2012, 2022-10-17:
Yes i have sent via github

Thanks

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

YifanWuBambu, 2022-10-17:
@P2012 
Thanks!

lanewei120, 2022-10-17:
> Hi i spoke to will @ (support) on your dicord and he told me to contact you. i have a problem using the slicer it takes at least 2 mins to load up will thinks it may be a network issue. i have no problems with other slicers. he told me to add the bambu config folder to help you
> 
> [image]

Hello,
after the Studio load up
is the network function workable?
any other issues there?

also please help to upload the log files to us from
C:\Users\user.name\AppData\Roaming\BambuStudio\log

thanks

lanewei120, 2022-10-17:
> 

1. have you tried printing?
it is very likely that your studio can not access network,
2. could you help to give some pictures of your error on studio? 
3. and also please help to upload the log files in C:\Users\user.name\AppData\Roaming\BambuStudio\log here
we can check the root cause then

P2012, 2022-10-17:
1. No not yet i just got my printer going to use it when i move to my new
house
2. Ok i will send to you,see attached
3. I had a look at this and it just says plugins and that is blank,see
attached

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

lanewei120, 2022-10-17:
> 1. No not yet i just got my printer going to use it when i move to my new house 2. Ok i will send to you,see attached 3. I had a look at this and it just says plugins and that is blank,see attached [...]

I am afraid I could not find your attached files
did you forget to upload?

P2012, 2022-10-17:
[image]
[image]

Hi see attached.yes i sent them but now trying github

P2012, 2022-10-17:
Sending via github

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

lanewei120, 2022-10-17:
very strange, it seems no plugins found in your directory
but the login status is on from your picture

1.  what is your BambuStudio version?
2.  could you remove the directory C:\Users\user.name\AppData\Roaming\BambuStudio and retry again
3.  and if it still fails, please help to pack and upload the log files in C:\Users\user.name\AppData\Roaming\BambuStudio\log here?

thanks

P2012, 2022-10-17:
> 1, Deleted it and then rebooted the system
> 2. Just downloaded the latest ver from the software site
> 3. It is the same as before
> 4. No log file just a plugin folder which is blank
>
>
> PAUL DANCE Director
> www.diamondhlifting.co.uk
> 0800 022 4475

Is it because i haven't put my printer on the network yet i have just
downloaded the software to have a play?

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

lanewei120, 2022-10-18:
> Is it because i haven't put my printer on the network yet i have just downloaded the software to have a play? PAUL DANCE Director
> […](#)
> ________________________________ [...]

no, whenever you connect to the printer or not, there should be lots of directories and files under
C:\Users\current_user_name\AppData\Roaming\BambuStudio

I guess it is just because your current user is not 'user'?
could you check your current windows user account
and goto the correct directory under the current user

it should be like this 
[image]

lanewei120, 2022-10-18:
you can also find the guick start of bambustudio for the first time from here
https://wiki.bambulab.com/en/software/bambu-studio/studio-quick-start

P2012, 2022-10-18:
Hi,

1. I have tried this and their is no log file in Bambu studio its just the
same as i sent you before

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

lanewei120, 2022-10-18:
> > Is it because i haven't put my printer on the network yet i have just downloaded the software to have a play? PAUL DANCE Director
> > […](#)
> 
> no, whenever you connect to the printer or not, there should be lots of directories and files under C:\Users\current_user_name\AppData\Roaming\BambuStudio
> 
> I guess it is just because your current user is not 'user'? could you check your current windows user account and goto the correct directory under the current user
> 
> it should be like this [image]

how about this?

P2012, 2022-10-18:
Please see attached there is no log file in Bambu studio

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

P2012, 2022-10-20:
Any news?
I also tried to download the new ver 1.3 and got this message pop up.see
picture attached.

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

P2012, 2022-10-20:
1.Hi i also had this message to see picture below

2.Is there anyway you could log on to my pc to sort this problem as it has
gone on to long now.

Thanks

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

lanewei120, 2022-10-21:
> 

Hi, I am sorry we were busy on the new version and didn't respond in time

we can not see the picture you  attached
it seems you missed it?

[image]

also, have you checked whether the current user name is correct or not?
```
I guess it is just because your current user is not 'user'?
could you check your current windows user account
and goto the correct directory under the current user
```

lanewei120, 2022-10-21:
> 1.Hi i also had this message to see picture below 2.Is there anyway you could log on to my pc to sort this problem as it has gone on to long now. Thanks PAUL DANCE Director
> […](#) [...]

there is some way to help to do this
could you send your discord id to my email lane.wei@bambulab.com?
then I will contact with you directly by discord or facebook

P2012, 2022-10-21:
Ok i sent you an email

lanewei120, 2022-10-21:
after debug, we found it is a issue caused by security software blocking,
now the issue has gone, close it

closed 2022-10-21 by lanewei120

P2012, 2022-10-21:
Great, thanks

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475