bambulab/BambuStudio #267: Feature - Live on-screen flow adjustment

DocHogan, 2022-09-13:
Very much needed. Live on-screen (at printer display, as well as in both Bambu Studio and Bambu Handy). Live on-screen flow rate adjustment (flow rate multiplier is likely best/easiest), to allow changing flow rate during print.  Now that textured PEI beds are officially supported and disable automated LIDAR flow calibration, as well as other times where automated flow calibration may fail (dirty lens, incompatible filament), there are times flow needs to be adjusted manually. Flow rates, volumes and multiplier can be set permanently in filament profile, but repeatedly setting values in profile and restarting print until optimal flow is achieved is a very wasteful (in filament and time) to dial in the correct value. What is needed is both a numeric input field and an incremental button set, available on the Repetier interface of the LCD, Bambu Studio, and Bambu Handy. No M500 to save settings, since this needs to be saved in filament profile; correct value can be noted by user and manually edited in filament profile in Studio. We just need a good way to FIND the value when LIDAR is not an option.

SaltWei, 2022-09-14:
Not familar with printer firmware, I just provide the method which I used.

M-K3y, 2022-09-30:
That would be a great thing, firstly you could use the whole space with the calibrated values and you could also use a filament for other printers which would calibrate the flow

WZ9V, 2022-12-04:
Would love a way to profile the filament and have the numbers displayed on the screen or in Handy or even sent to Studio so I can update the filament settings in the Slicer.  I like the Lidar calibration but I'd prefer to do it once for a filament and not each print.