bambulab/BambuStudio #269: Autoarrange bug in 01.02.00.99

AnallSafari, 2022-09-13:
There appears to be an auto arrange bug with multiple parts.

Steps to reproduce:

Add this stl to slicer:
[Parts_Tray_v2.zip](https://github.com/bambulab/BambuStudio/files/9561493/Parts_Tray_v2.zip)
Select part and control c
Paste in 5 new models.
Arrange.

It will arrange 3 of the trays in the middle and another 3 off screen.

Manually arranging the tray and then using the arrange function produces the same result.

Here is a capture using Windows game mode, please note that it doesnt capture the rightclick -->arrange 
https://user-images.githubusercontent.com/64895918/190029606-6c71882a-6a77-4970-9900-9e206a3e2cd1.mp4

labatt, 2022-09-14:
I noticed the same behavior yesterday. Please let me know if you'd like more details.

AnallSafari, 2022-09-15:
> I noticed the same behavior yesterday. Please let me know if you'd like more details.

Maybe upload the STL?

I did a print with really small parts and it had no issue arranging them on the bed, so it appears to be a size issue.

tangketan, 2022-09-17:
I've tried the stl's and the auto-arranging is fine. Have you tried different options?
[image]

closed 2022-11-09 by tangketan