bambulab/BambuStudio #270: AUX fan not turning on after first layer in 01.02.00.99

adamhf, 2022-09-14:
When printing PLA in v1.1.1.1 the AUX fan would be enabled after printing and validating the first layer.  In 01.02.00.99 the AUX fan is enabled during the first layer validation, but is disabled after it has been completed and is not re-enabled during the print.

SaltWei, 2022-09-15:
Can't reproduce, please share the 3MF or check whether the filament preset you used disables the AUX fan.
[image]

adamhf, 2022-09-15:
Hey @SaltWei,

I appreciate you getting back to me.

I have confirmed that 

the Aux part cooling fan is enabled in the filament profile, its the default Bambu Matte PLA profile.

[image]

Here is a 3MF exported from 01.02.00.99 running on Windows 11

[aux-fan-test.zip](https://github.com/bambulab/BambuStudio/files/9577507/aux-fan-test.zip)

adamhf, 2022-09-15:
Here is the same model sliced in 1.1.1.1 with the AUX fan working after layer 1.

[image]

[aux-fan-test-working.3mf.zip](https://github.com/bambulab/BambuStudio/files/9577684/aux-fan-test-working.3mf.zip)

adamhf, 2022-09-15:
I did some analysis with mowcius on the Discord and it looks like the ordering of the `M976 S1 P1` command in the gcode has changed, presumably in [this MR](https://github.com/bambulab/BambuStudio/commit/e7c977f86b1a727541f20eeec18699abef46c2ce).

Interestingly if I disable the First Layer Inspection the AUX fan spin up as expected

SaltWei, 2022-09-16:
Yes. The aux fan is opened in gcode by M976 P2 S178.
I have comfirmed with printer firmware guys, and this is bug of printer firmware and will be fixed soon.
Thank you very much for this finding!

adamhf, 2022-09-16:
Not a problem :) thanks for the update.

closed 2022-10-12 by YifanWuBambu