bambulab/BambuStudio #278: Feature Request - Multi color / Multi material without AMS or in conjunction with AMS

Printman76, 2022-09-14:
Goal - I want to print a model that will have a PLA body with TPU feet.

1) I would like to be able to slice the model with the 2 materials (Both will be manually loaded - No AMS)
- The 1st Material will be loaded (PLA) and when the print gets to the height of material change on screen prompts will guide through the change of material and print will resume ( This will take into account material temperature and flow rate set for material in slicer)

2) Same situation but using AMS for 1 material and manual feed of 2nd (TPU)
    - This would be facilitated with the Y adapter https://www.printables.com/model/254250-bowden-y-splitter-for-bambu-lab-x1
    - I would select a material in the AMS and a second that is not in the AMS 
    - I would be prompted at the time of sending the print to the printer with a popup window explaining that 1 of the colors/materials would need manual loading as it is outside of the AMS selections and that the Y adapter is needed

parksddd, 2022-09-21:
@Printman76 I've been able to get this to work manually, but I agree, a more automated feature would be very welcome.

Currently my process without AMS is to add a pause via slicer to the layer that needs the swap.
On the device click the "close" button for the pause, then navigate to the settings page and manually set an unload temperature.
Once temp is reached I reverse the extruder until the filament and manually unload it. This, unfortunately, does not work with the filament cutting feature as the unload button is not functional during the pause.
I then put in new filament and extrude until the color changes/etc.
Navigate back to the print page and click resume.
Workable but not the best.

My minimal ask for the above would be the ability to mark a layer as a filament change and have it automate the unload of the existing filament. Then upon 'resume' run a purge and continue.

Gr4sil, 2023-03-03:
I would want to make filament changes within middle of the layer. Means that I could make signs etc. by make the changes by myself without AMS.

Feature should work like with the AMS, but you would have prompts within P1P and X1 ”Unload filament 1” ”Load filament 2”.

EukaDesigns, 2023-03-09:
@Printman76 @parksddd 
I've written some g-code to unload and pause for the filament change. This only works if the AMS is disconnected, and filament loads are manual. This g-code replaces the g-code for filament change in the printer settings. The filament cut/retract motions were my estimations of what the P1P does when clicking the unload button. I'm hoping to get the actual g-code from BambuLabs, but for now this has worked for me. 

G-code here: [change_filament_noAMS.gcode](https://github.com/EukaDesigns/Bambu_CustomGCode/blob/bfdd8d4e9dce94def22d21b0e4ac8b5e49ebadeb/change_filament_noAMS.gcode)

ab3dgeek, 2023-03-09:
oh yes, manual layer change at a given layer would be nice, doing it at specific moments within the layer would be even better. I would love to try out a few things with pla and tpu printed in the same layers.

EukaDesigns, 2023-03-09:
The change_filament_noAMS code should work for any filament changes, including if you've painted it in BambuStudio rather than set layer heights.

adripo, 2023-07-18:
Thank you @EukaDesigns! I see that the gcode in this repo is now different from the original one from your repo. Can you create an updated version, please?

adripo, 2023-07-18:
I created a new version based on the updated g-code, and also corrected a couple of things.
Here is it:
https://gist.github.com/adripo/4d0df406ea86de9d1a52f09dab318538

druckgott, 2024-09-05:
Is there a way to included this
https://github.com/applenana/AP-AMS
In comnination with original ams this Would be realy cool ams all But no tpu and one automatik point over the regulär Manual load with tpu