bambulab/BambuStudio #279: Feature Request: Add delay entry for end of time-lapse

JPotter73, 2022-09-14:
I suggest a text box entry for an added delay in seconds to add to the end of the time-lapse for presentation. Or perhaps end frame counts to add based on the time of the final layer. Either that or hard code the delay to something that works nicely

Right now it seems the end of the print resets to the start before being able to enjoy the final product.

YifanWuBambu, 2022-09-17:
@JPotter73 
That's a good idea. We will add it in the next release.

dereknheiley, 2022-09-17:
> I suggest a text box entry for an added delay in seconds to add to the end of the time-lapse for presentation. Or perhaps end frame counts to add based on the time of the final layer. Either that or hard code the delay to something that works nicely
> 
> Right now it seems the end of the time-lapse resets to the start before being able to enjoy the final product.

I think you can manually do this in the machine end gcode

JPotter73, 2022-09-17:
Thanks dereknheiley. I may look into this. Right now, I'm not printing anything exciting to worry about it, it was just an observation that may make it more accessible by the masses. I would need a better understanding of the time-lapse triggering as I think it's based on layer change. Adding delays with blank layers might be a hacky way for me to do this.