bambulab/BambuStudio #281: Feature Request: Use LiDAR to calibrate maximum volumetric flow rate

Losi-GL, 2022-09-14:
Hello everybody,
To preface, I am currently working on a SuperSlicer profile for the X1 that uses the "Autospeed" function to set a base speed from the maximum volumetric flow rate then it modifies that base speed as percentages for all moves. Essentially this automatically sets all speeds for me based on my layer height, line width, and the maximum flow of the filament (note in Bambu Slicer if using a slow flowing filament and high layer height, it is possible to have the entire print go at the same speed because it is limited by the maximum volumetric rate. In my design the speeds are all adjusted off of the filament's characteristics.) This is similar to the feature request listed here -https://github.com/bambulab/BambuStudio/issues/126

While going through the start g-code I realized that if Bambu simply draws a high volumetric demand line (unique for each nozzle), let's say a 0.3-layer height x 0.5-line width for a 0.4 nozzle, then simply moves the print head faster and faster to increase the volumetric demand, the LiDAR should be able to "see" where the line width begins to decrease and correspond that with a maximum volumetric flow rate, which would then in turn set most speeds (It may be worthwhile to consider volumetric marlin... instead of trying to match the flow to the movements of the nozzle, match the movements of the nozzle to the flow, based around the maximum calibrated flow rate)

GuyH77, 2022-09-15:
> Hello everybody,
>
> To preface, I am currently working on a SuperSlicer profile for the X1 that uses the "Autospeed" function to set a base speed from the maximum volumetric flow rate then it modifies [...]

Seems a great idea to me

Losi-GL, 2023-08-02:
I just wanted to drop a note to recognize the awesome work the Bambu team is doing. Seeing this begin implementation on the X1 in Firmware 1.06 is most excellent! I hope all those involved in this are proud to implement features that continue to push 3D printing further. Thank you guys for building this!