bambulab/BambuStudio #286: Feature Request - Print with no outer wall / infill only

KuraiKitsune, 2022-09-16:
I would like the ability to print spongy TPU structures with the Bambu X1. This is without having an outer wall, so the whole model is printed with just infill structure.

In the purpose case I've seen possible on Prusa's slicer software, an outer wall ideally made of denser infill pattern, with a less dense internal pattern.

Screenshot attached.
[image]

tangketan, 2022-09-17:
Seems making min value of wall count to 0 will just do.

YifanWuBambu, 2022-09-18:
@KuraiKitsune 
The wall count is limited to larger than 1 currently. We will allow it to be 0 in the next release.

KuraiKitsune, 2022-09-19:
Any idea on printing an outer wall out of more "infill"?
Or maybe the option to have 2 objects with different print settings on a single project?

YifanWuBambu, 2022-09-21:
> 

@KuraiKitsune 
You may set per-object parameter by switching from "Global" to "Objects" mode on sidebar.
[image]

User manul provides detailed introductions.
[image]

closed 2022-09-21 by YifanWuBambu

SaltWei, 2022-09-21:
Done. Please wait for the next release

YifanWuBambu, 2022-10-21:
Added in v1.3.0.12.

hella909, 2023-05-29:
> Any idea on printing an outer wall out of more "infill"? Or maybe the option to have 2 objects with different print settings on a single project?

So far, I was able to place slightly overlapping original design modifiers and set significantly denser infill on them. Say, if your design is rectangular, you will have to place 4 modifiers on all 4 sides. Nothing special, just smaller narrow rectangles. For unknown to me reason in my case I was able to produce much denser infill only on 2 out of 4 sides, but for me it was sufficient enough. Hope it'll help!

hella909, 2023-05-29:
> So far, I was able to place slightly overlapping original design modifiers and set significantly denser infill on them. Say, if your design is rectangular, you will have to place 4 modifiers on all [...]

To be continued... :)
My bad! Just noticed that 2 of the modifiers were floating over the design, it's why I couldn't get all 4 sides print with denser infill. Problem solved! All 4 sides are printed as should, very dense, while the central part's infill is loose.

cwale, 2024-01-18:
I believe this functionality may now no longer work? I'm trying to use modifiers to create holes in a part, but walls are still generated. Please see attached. I was expecting that I would be able to see holes in the 'Main Body' and directly into the infill, i.e. no wall loop on the modified section.
[image]
[image]

BAMBU4U, 2024-10-16:
> Seems making min value of wall count to 0 will just do.

YES that works! You also have to ZERO TOP SHELL LAYERS and BOTTOM
[image]