bambulab/BambuStudio #287: Feature request - Add an option for an ooze shield

mmontminy, 2022-09-16:
Even with the purge chute and a prime tower, otherwise perfect multi-material prints are often flawed due to how much the extruder oozes.

I just did a print that I used tree supports on where the supports encased the majority of the lower half of the print, effectively acting as an ooze shield. The lower half of the print was perfect. The upper half where there was no support had a few areas where oozed filament had bonded to the print.

While it looks possible to use a skirt as tall as the print I suspect this wouldn't have enough adhesion for tall prints.

Please consider adding a proper ooze shield as an option.

tangketan, 2022-09-17:
Thanks for the suggestion. But before adding the shield, we first need to investigate why prime tower doesn't work well as it should be.

bani6809, 2022-11-07:
i'm baffled why a prime tower is even needed when we have a purge chute and fast corexy.  prime tower is just an additional point of failure for a print.

kiliankoe, 2022-11-07:
@bani6809 It's in the name, for priming. After purging the nozzle needs to be primed or else you'll get a bunch of under-extrusion issues where the nozzle starts printing. You can easily see this if you manually delete your prime tower.

bani6809, 2022-11-07:
@kiliankoe i have actually tested this though. i deleted the prime tower and added a thorough prime + multiple nozzle wipe and basically prime tower is simply not needed, it's antiquated and other printers with similar capabilities are doing away with prime towers also. i believe 3d print general did a video recently on a printer which did fine with multi filament print without one, to his suprise.

suit, 2025-08-30:
If you’ve stumbled upon this and are wondering about ooze shields: in theory, they’re a great idea — and that got me thinking.

Why do we wipe the nozzle and use a prime tower in the first place? To stabilize pressure in the nozzle and to catch the stray whiskers of filament that ooze out during a nozzle switch (something that’s especially relevant for the H2D).

Back in the early days of FDM printing, there was no such thing as a purge line. Instead, we relied on the skirt loop — a feature that’s still available in slicers today. And it actually does exactly what we need: it can replace the prime tower and act as a shield at the same time. All you have to do is extend the skirt loops all the way up, layer by layer, until the final filament change.

I tested this by printing a two-color Benchy (intentionally at too high a temperature to maximize oozing). The results? It kind of works. The skirt traps many of the long whiskers, but since it only prints loops with the currently loaded filament — and not right after a filament switch — it doesn’t catch everything.

Even though the print looks pretty rough, the point is clear: an ooze shield could absolutely work if implemented properly. What’s needed is a slicer feature that generates one loop for each color on every layer, and crucially, prints the loop immediately after a nozzle switch rather than at the start of the layer. That way, it would serve the same function as the purge tower, but more efficiently.

[image]
[image]