bambulab/BambuStudio #289: [Feature Request] AMS Random &  Pattern Color Mode

awrefai, 2022-09-17:
Hi,

It would be nice to add a feature to randomly print from AMS available filaments or with a specific pattern. This feature is similar to Mosaic Palette where a random color is printed in a non-specified order or in a specific pattern to create a nice print without manually painting in the slicer

Thanks

YifanWuBambu, 2022-09-18:
Hi @awrefai ,
Could you show me a picture of the feature? Thanks!

awrefai, 2022-09-19:
Hi @YifanWuBambu,

Please find the following samples of the feature. 
[image]

[image]

[image]

YifanWuBambu, 2022-09-21:
@awrefai 
Thanks for the example.
We have added this to our plan list.

awrefai, 2022-09-21:
Thank you! @YifanWuBambu