bambulab/BambuStudio #290: When rescaling an object "Uniform scale" is checked again

coat64, 2022-09-17:
When you scale an object and unchecked "Uniform Scale", it is enabled again when you rescale the same object.  Annoying when you are tweaking an object.

YifanWuBambu, 2022-09-18:
Hi @coat64 ,
What's your studio version? I tested on v1.2.0.9, but did not reproduce this issue.
[image]

coat64, 2022-09-19:
I use 1.2.0.9, the latest version (windows 11, amd).

Yathani, 2022-09-19:
I have the same issue .. uniform scale does not work on Z axis scaling

YifanWuBambu, 2022-09-21:
@coat64 
Deselect and re-select an object, then reproduced this issue.
We will fix the bug.

YifanWuBambu, 2022-09-21:
@Yathani 
Do you change z value in the text input box? Or by dragging the blue cubic grabber in 3D scene?
[image]

Yathani, 2022-09-21:
> @Yathani Do you change z value in the text input box? Or by dragging the blue cubic grabber in 3D scene? [image]

Changing the Z value directly does scale/descale properly but using the cube does not. This is applies to all axis X, Y and X.

This issue only happen when using the Blue, Green or Red cube for scaling, the turquoise/light blue corner cube works fine.

coat64, 2022-09-21:
Maybe we misunderstood each other :) or you missed the line at the bottom. The small push-pull-cubes works fine. But if you have an object, go to scale, uncheck "uniform scale) enter a value(s) in one or more of the boxes, the object deforms as expected. When you want to do it again for the same or other value, the checkbox is checked again. This is annoying, if you forget to uncheck it first (and you will), change a value, the other two axis are also reset and you can start over.

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

coat64, 2022-10-22:
Yes  I saw :). Thanks for listening to user-input.