bambulab/BambuStudio #292: Adding Arch based Distros to Linux Buildscript

lord-x-type, 2022-09-18:
Hi, after trying to compile the slicer i am stuck on some problems as seen in #236 
i have seen the BuildLinux.sh but this script is for Debian based Distributions like Ubuntu.
Would it be possible to add Arch based Distros like Manjaro etc to the script?
I tried to change some of the code to run on Manjaro but without big success.
this would be very nice :)
Thanks alot

deftdawg, 2022-09-23:
Fixing this would not be a good use of development resources for 2 reasons:

1) The output of the build process, an AppImage, should run on any distro

2) Any distro including Arch should already be able to build that AppImage using a Podman container using the `Containerfile` in the root of the repository... The same file *might* also be usable with Docker

My recommendation is build in Podman and report any issues you have with that or the AppImage it produces.

lord-x-type, 2022-09-23:
Thanks for your reply.
You are absolutely right.. didn´t saw it that way.
This Request can be closed.

closed 2022-09-23 by lord-x-type