bambulab/BambuStudio #293: Invalid map error

LegacyOfBob, 2022-09-18:
Sliced a file and this error message popped up:
"invalid map<K, T> key"

Bambu Studio now refuses to slice anything, needs to be restarted to resume working. 

[image]

SaltWei, 2022-09-20:
Could you please share the 3MF file for us to reproduce and debug ?

LegacyOfBob, 2022-09-21:
Here is the 3mf file
https://drive.google.com/file/d/1PBbigty6BuD49wXxeE3QZOf6SQxPG-hN/view?usp=sharing

ChrisG-hub, 2022-09-21:
I got no issues opening the file. Must be a local issue.

[image]

zengzm1212, 2022-09-26:
@LegacyOfBob  I open the file and slice successfully. Could you provide your software version uesd to open the file.

SaltWei, 2022-12-16:
Can't reproduce this issue and keep watching for a long time.
Close.
If it appears again, please report new ticket to us. Thank you very much!

closed 2022-12-16 by SaltWei