bambulab/BambuStudio #294: Bambu slicer bug

P2012, 2022-09-18:
[image]
On the home screen you see this as pic when i click on anyone of them i have an error message appear error 1155

bambu123, 2022-09-21:
Can you open urls such as "https://github.com/bambulab/BambuStudio/issues/294" from Windows Startup？
[image]

P2012, 2022-09-21:
[image]
My startup folder is empty

bambu123, 2022-09-22:
No, it is not the startup folder.
I mean the Windows Run Dialog as shown previously.
You press Win + R， to invoke this dialog， and then input an HTTP URL to open. Can it be opened?

P2012, 2022-09-23:
Hi i done this and it just makes the screen bigger.

PAUL DANCE Director
________________________________
www.diamondhlifting.co. <http://www.diamondhlifting.co.uk>uk
0800 022 4475

bambu123, 2022-09-27:
I think there are no web browsers that can process HTTPS URLs on your computer.
Have you installed chrome?

P2012, 2022-09-27:
Hi no i am using normal google shall i install chrome?

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

bambu123, 2023-02-02:
@P2012 Have the problem been solved?

P2012, 2023-02-02:
Yes all working

Thanks

PAUL DANCE Director
________________________________
www.diamondhlifting.co.uk
0800 022 4475

closed 2023-03-02 by SaltWei